ros2/geometry2 #310: Add WithCovarianceStamped message support to tf2_geometry_msgs

JWhitleyWork, 2020-08-13:
## Feature request

#### Feature description
Especially in localization systems, covariance is king. Unfortunately, there are no convenience features to transform a WithCovarianceStamped-based message in `tf2_geometry_msgs` from one frame to another (only the Stamped versions). The only relevant message of this type that I care about at the moment is PoseWithCovarianceStamped but others might be worth discussing.

#### Implementation considerations
Should be fairly easy. Working on implementation now but would love to get feedback.

clalancette, 2020-08-14:
It seems reasonable to me.  @tfoote any thoughts?

tfoote, 2020-08-14:
Yeah, covariances should be able to have canonical transformations. A contribution to extend the transform datatypes would be appreciated.

JWhitleyWork, 2020-08-17:
@tfoote Question related to this: There are currently functions in `tf2_geometry_msgs` which imply the following equivalencies:

- `geometry_msgs::msg::Transform` == `tf2::Transform`
- `geometry_msgs::msg::TransformStamped` == `tf2::Stamped<tf2::Transform>'
- `geometry_msgs::msg::Pose` == `tf2::Transform`

However, the following equivalencies are missing:

- `geometry_msgs::msg::PoseStamped` == `tf2::Stamped<tf2::Transform>`

Was this intentional?

JWhitleyWork, 2020-08-17:
Nevermind. It looks like creating that last equivalency isn't possible because of how `tf2_geometry_msgs` utilizes template deduction. I'll just define the `doTransform` function and be done with it.

JWhitleyWork, 2020-08-31:
@clalancette and/or @tfoote - Can one of you have a look at #312?

surfertas, 2020-09-15:
@clalancette Kind reminder to review if this can be closed.

clalancette, 2020-09-15:
> @clalancette Kind reminder to review if this can be closed.

Thanks for the ping; now that #312 is merged, I think we can close this.

closed 2020-09-15 by clalancette