ros2/geometry2 #316: Reconsider the function signature of `set_transform()` and `set_transform_static()`

surfertas, 2020-09-10:
@sloretz Should the signature of `set_transform()` and `set_transform_static()` be respectfully:

```python
def set_transform(self, *args):
    ...

def set_transform_static(self, *args):
    ...
```

I ask as I see that the function signature at https://github.com/ros2/geometry2/blob/a4f152f67d09b50d6d6450bb00c44512e52a880c/tf2_py/src/tf2_py.cpp#L656 and also see that `PyArg_ParseTuple()` is used as opposed to `PyArg_ParseTupleAndKeywords()`

_Originally posted by @surfertas in https://github.com/ros2/geometry2/pull/315#discussion_r483688462_

sloretz, 2020-09-11:
> and also see that PyArg_ParseTuple() is used as opposed to PyArg_ParseTupleAndKeywords()

Yeah, I think you're right that there doesn't need to be a `**kwargs`. [It's only using positional arguments](https://docs.python.org/3/c-api/arg.html#c.PyArg_ParseTuple). I assume it's using `*args, **kwargs` to pass through args without duplicating knowledge of what `set_transform()` and `set_transform_static()` take, but if there are now type hints I think that's no longer relevant. With the type hints I would recommend changing the signatures to show the true arguments instead of using `*args: Tuple[...]`.

```python
def set_transform(self, transform: TransformStamped, authority: str):
    ...

def set_transform_static(self, transform: TransformStamped, authority: str):
    ...
```

surfertas, 2020-09-12:
@sloretz Noted. Will take care of this and create a PR.

PR #321 (2020-09-14): Changed the signatures of set_transform and set_transform_static to show true arguments.
@clalancette @sloretz Changed the signatures of `set_transform` and `set_transform_static` to show true arguments instead of using *args: Tuple[...].

closes #316
closed 2020-09-14 by sloretz