ros2/geometry2 #329: tf2_py build error when building ROS2 foxy from source

Michael-Equi, 2020-10-09:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - Source
- Version or commit hash:
  - 64d46d6

#### Steps to reproduce issue
Follow the instructions [here](https://index.ros.org/doc/ros2/Installation/Foxy/Linux-Development-Setup/) on ubuntu 18.04 (in my case using a Jetson Nano).

#### Expected behavior
Colcon build should build tf2_py successfully. 

#### Actual behavior
All packages build fine except for tf2_py and one of its dependencies. 

#### Additional information
Here is the error 

```
Starting >>> tf2_py  
--- stderr: tf2_py                         
CMake Error at /opt/ros/foxy/install/share/tf2/cmake/ament_cmake_export_targets-extras.cmake:18 (message):
  Failed to find exported target names in
  '/opt/ros/foxy/install/share/tf2/cmake/tf2Export.cmake'
Call Stack (most recent call first):
  /opt/ros/foxy/install/share/tf2/cmake/tf2Config.cmake:41 (include)
  CMakeLists.txt:26 (find_package)

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/colcon_core/executor/__init__.py", line 91, in __call__
    rc = await self.task(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/colcon_core/task/__init__.py", line 93, in __call__
    return await task_method(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/colcon_ros/task/ament_cmake/build.py", line 66, in build
    if await has_target(args.build_base, 'install'):
  File "/usr/lib/python3/dist-packages/colcon_cmake/task/cmake/__init__.py", line 57, in has_target
    return target in await get_makefile_targets(path)
  File "/usr/lib/python3/dist-packages/colcon_cmake/task/cmake/__init__.py", line 78, in get_makefile_targets
    CMAKE_EXECUTABLE, '--build', path, '--target', 'help'], cwd=path)
  File "/usr/lib/python3/dist-packages/colcon_core/subprocess.py", line 110, in check_output
    'Expected {args} to pass: {stderr_data}'.format_map(locals())
AssertionError: Expected ['/usr/bin/cmake', '--build', '/opt/ros/foxy/build/tf2_py', '--target', 'help'] to pass: make: *** No rule to make target 'help'.  Stop.

---
Failed   <<< tf2_py [2.56s, exited with code 1]
```

clalancette, 2020-10-12:
Thanks for the report.

I think this is another instance of https://github.com/ros2/console_bridge_vendor/pull/25 .  @ahcorde would you mind doing a backport to Foxy?

ahcorde, 2020-10-12:
Backported https://github.com/ros2/console_bridge_vendor/pull/26

Michael-Equi, 2020-10-15:
I pulled the latest version of console_bridge_vendor and rebuilt tf2_py. I am still getting the same build error.

Michael-Equi, 2020-10-16:
When I checkout the eloquent branch of this repository tf2_py builds fine.

clalancette, 2020-10-16:
For what its worth, this builds fine for me on 18.04 (amd64 in my case).  Have you tried cleaning your workspace (`rm -rf build install log`) and building again?

Michael-Equi, 2020-10-16:
That worked, I think I had to rebuild tf2 and tf2_ros to again specifically.

closed 2020-10-16 by Michael-Equi

clalancette, 2020-10-16:
> That worked, I think I had to rebuild tf2 and tf2_ros to again specifically.

Great, thanks for retesting.