ros2/geometry2 #337: Any plans to support doTransform() with non-stamped message types?

mitsudome-r, 2020-10-30:
## Feature request
In ROS1, we used to have doTransforms for message types without Headers, but they are missing in ROS2.
Are there any reasons why will omitted these functions in ROS2?

Example with geometry_msgs/Point
[ROS1](https://github.com/ros/geometry2/blob/noetic-devel/tf2_geometry_msgs/include/tf2_geometry_msgs/tf2_geometry_msgs.h#L722) 
[ROS2](https://github.com/ros2/geometry2/blob/ce8f0070b08389c6c8c0874c4eb1c6143d893bb6/tf2_geometry_msgs/include/tf2_geometry_msgs/tf2_geometry_msgs.h#L158)

#### Feature description
Transforms geometry messages without timestamps.

#### Implementation considerations

clalancette, 2020-10-30:
The original commit that added those to https://github.com/ros/geometry2 is [here](https://github.com/ros/geometry2/commit/21e2d9378742116b9ccb7d2e95ed0963e221b231).  That was added after we forked this repository off from that one.

No fundamental objection to having those.  Contributions to add it are very welcome.

adamkrawczyk, 2021-09-03:
Hi,

I'm doing this and create PR but wondering whether adding PoseWithCovariance without timestamp makes sense? 
 
Best

aprotyas, 2021-09-03:
> wondering whether adding PoseWithCovariance without timestamp makes sense?

You don't necessarily need headers for all types of applications. For example, I can think of a planar path planner that would not care much about the timestamp/reference frame of goal queries expressed as `PoseWithCovariance`s.

adamkrawczyk, 2021-09-03:
Made changes at:
https://github.com/ros2/geometry2/pull/452

Please take a look and share your thoughts

aprotyas, 2021-10-15:
@galou I might be misunderstanding the math here, but I'm thinking that if you do a transformation on a stamped pose, then the output of that transformation should be in the same frame as the transformation (and not the input pose), which is what's happening in the current implementation of `doTransform<geometry_msgs::msg::PoseStamped>`

CursedRock17, 2024-10-26:
Just to double check, I think this issue was resolved by the matching PR, but does it still need help in any sort of way?