ros2/geometry2 #338: Buffer::lookupTransform hangs with sim time clock

rrrand, 2020-10-30:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - binaries
- Version or commit hash:
  - foxy

#### Steps to reproduce issue
Start gazebo and provide parameter use_sim_time value true. Create `tf2_ros::Buffer` with `my_node->get_clock()`.
Then `lookupTransform` hangs because of time check implementation in `canTransform` (https://github.com/ros2/geometry2/blob/ros2/tf2_ros/src/buffer.cpp#L183).

There is better example in https://answers.ros.org/question/307682/how-to-set-up-tfbuffer-clock-in-ros-2-so-it-works-with-sim-time/

#### Additional information

1. lookupTransform does not freeze with zero timeout.
2. I am not sure that provided behaviour is an issue. In my projects I construct Buffer with system clocks. So it is not a problem for me now. But there are projects that use rclcpp::Node::get_clock (e.g. https://github.com/introlab/rtabmap_ros/blob/ros2/src/CoreWrapper.cpp#L132). Such projects hang with gazebo. And may be it is their problem. However, imo, if there are problems in several projects, it is better to solve it here. At least giving some words in constructor comments other than ` * \param clock A clock to use for time and sleeping`.
3. Learning sources of Buffer I found that in lookupTransform method canTransform is called but its return value is not used (https://github.com/ros2/geometry2/blob/ros2/tf2_ros/src/buffer.cpp#L101). Is it correct?

tfoote, 2020-10-30:
This will block, but this is doing exactly what you're asking it to do. You're calling a blocking call inside of a specifically constructed single threaded process. If you are forcing things to run single threaded you do need to be careful about knowing what is blocking. You can call the non-blocking version as mentioned in 1) or you can provide a parallel thread to spin and service the other potential interactions. 

2) That's strongly not recommended because you will not be able to take advantage of simulated time. There's ongoing efforts to make constructing the clock better. But it's highly recommended to just ask for the clock from the Node instance which is already wired up.

3) Yes that's correct. That call is optimized to not generate results repeatedly and quickly during the polling period and then the same parameters are passed to lookupTransform which will generate the full results and return them or the equivalent error message. Subject to one more retry in the case it hasn't worked yet.

blacksoul000, 2020-11-02:
Maybe at least log a warning? Wrong use of this code is so easy =(
or You can do some checks to avoid it.

clalancette, 2020-11-18:
I agree that use of this code is somewhat hard to get right.  But it is also hard to log a warning, since you need to know not only that you are in a callback, you also need to know which executor and callback group you are in (we *should* log a warning for the single-threaded executor but we shouldn't necessarily warn for the multi-threaded executor).

I'm going to mark this as "help wanted"; it would be nice to make it easier to use, its just not totally clear how to do that at the moment.

peci1, 2025-04-09:
Maybe an easy solution: the buffer could check whether `setTransform()` and `canTransform()` are being called from the same thread (e.g. by saving the thread_id or by keeping a pair of thread_local variables). Would that make sense?

I know there may be use-cases when this is legal (e.g. unit tests or prefilled buffers). But the general case could be this simple, couldn't it?

peci1, 2025-04-09:
Hmm, but this might work wrong on a multithreaded executor which could actually schedule the transform setting and getting to the same thread some time after each other...