ros2/geometry2 #341: cmake 3.18.4 compile error regresseion

AchmadFathoni, 2020-11-08:
## Bug report

**Required Info:**

- Operating System: 
  - Arch
- Installation type:
  - source
- Version or commit hash:
  - 0.13.6

#### Steps to reproduce issue
Compiling tf2_eigen, tf2_geometry_msgs, or tf2_kdl using CMake 3.18.4

#### Expected behavior
Successfully compiling.
#### Actual behavior
```
Starting >>> tf2_geometry_msgs
--- stderr: tf2_geometry_msgs                         
CMake Error at /usr/share/cmake-3.18/Modules/FindPackageHandleStandardArgs.cmake:165 (message):
  Could NOT find Threads (missing: Threads_FOUND)
Call Stack (most recent call first):
  /usr/share/cmake-3.18/Modules/FindPackageHandleStandardArgs.cmake:458 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake-3.18/Modules/FindThreads.cmake:234 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  /home/toni/Documents/source/ros2_foxy/install/rmw_implementation/share/rmw_implementation/cmake/rmw_implementation-extras.cmake:67 (find_package)
  /home/toni/Documents/source/ros2_foxy/install/rmw_implementation/share/rmw_implementation/cmake/rmw_implementationConfig.cmake:41 (include)
  /home/toni/Documents/source/ros2_foxy/install/rcl/share/rcl/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /home/toni/Documents/source/ros2_foxy/install/rcl/share/rcl/cmake/rclConfig.cmake:41 (include)
  /home/toni/Documents/source/ros2_foxy/install/libstatistics_collector/share/libstatistics_collector/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /home/toni/Documents/source/ros2_foxy/install/libstatistics_collector/share/libstatistics_collector/cmake/libstatistics_collectorConfig.cmake:41 (include)
  /home/toni/Documents/source/ros2_foxy/install/rclcpp/share/rclcpp/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /home/toni/Documents/source/ros2_foxy/install/rclcpp/share/rclcpp/cmake/rclcppConfig.cmake:41 (include)
  /home/toni/Documents/source/ros2_foxy/install/tf2_ros/share/tf2_ros/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /home/toni/Documents/source/ros2_foxy/install/tf2_ros/share/tf2_ros/cmake/tf2_rosConfig.cmake:41 (include)
  /home/toni/Documents/source/ros2_foxy/install/ament_cmake_auto/share/ament_cmake_auto/cmake/ament_auto_find_build_dependencies.cmake:67 (find_package)
  CMakeLists.txt:19 (ament_auto_find_build_dependencies)

---
Failed   <<< tf2_geometry_msgs [1.76s, exited with code 1]

Summary: 0 packages finished [2.57s]
  1 package failed: tf2_geometry_msgs
  1 package had stderr output: tf2_geometry_msgs
```
#### Additional information
Adding `set(CMAKE_THREAD_LIBS_INIT "-lpthread")` to each CMakeLists.txt fix the problem. Unfortunately I can't downgrade my CMake to confirm this regression. I also can't remember the last working CMake version I use before upgrading to current version.

philleer, 2023-10-24:
Anyone can give some suggestions? I have encouter the same issue.

philleer, 2023-10-24:
> > Anyone can give some useful suggestions? I have encoutered the same issue.
> 
> Have you read "Additional Information" above?

Yes, I saw it. But why?