ros2/geometry2 #342: Build error while using ros-foxy-tf2-geometry-msgs 0.13.6-1focal.20201103.15490

ashwinsushil, 2020-11-10:
## Bug report

I am using tf2-geometry-msgs in one of my projects. Upon building using `colcon build`, I get the error show in `Actual behavior`
```
#include <tf2_geometry_msgs/tf2_geometry_msgs.h>
```

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - binaries
- Version or commit hash:
  - 0.13.6-1focal.20201103.15490  (ROS2 foxy)
- DDS implementation:
  - Cyclone
- Client library (if applicable):
  - rclccp

#### Steps to reproduce issue
Build a ROS2 foxy package using tg2-geometry-msgs

#### Expected behavior
Succesful build.

#### Actual behavior
```
In file included from /umd2_ws/src/test/test_core/plugins/actions/navigate.cpp:19:
/opt/ros/foxy/include/tf2_geometry_msgs/tf2_geometry_msgs.h:287:114: error: ‘getCovarianceMatrix’ is not a template function
  287 |   std::array<std::array<double, 6>, 6> getCovarianceMatrix(const geometry_msgs::msg::PoseWithCovarianceStamped& t)  {return covarianceRowMajorToNested(t.pose.covariance);}
      |                                                                                                                  ^
/opt/ros/foxy/include/tf2_geometry_msgs/tf2_geometry_msgs.h:341:64: error: ‘WithCovarianceStamped’ in namespace ‘tf2’ does not name a template type
  341 | geometry_msgs::msg::PoseWithCovarianceStamped toMsg(const tf2::WithCovarianceStamped<tf2::Transform>& in)
      |                                                                ^~~~~~~~~~~~~~~~~~~~~
/opt/ros/foxy/include/tf2_geometry_msgs/tf2_geometry_msgs.h:341:85: error: expected ‘,’ or ‘...’ before ‘<’ token
  341 | geometry_msgs::msg::PoseWithCovarianceStamped toMsg(const tf2::WithCovarianceStamped<tf2::Transform>& in)
      |                                                                                     ^
/opt/ros/foxy/include/tf2_geometry_msgs/tf2_geometry_msgs.h:341:47: error: template-id ‘toMsg<>’ for ‘geometry_msgs::msg::PoseWithCovarianceStamped tf2::toMsg(int)’ does not match any template declaration
  341 | geometry_msgs::msg::PoseWithCovarianceStamped toMsg(const tf2::WithCovarianceStamped<tf2::Transform>& in)
      |                                               ^~~~~
/opt/ros/foxy/include/tf2_geometry_msgs/tf2_geometry_msgs.h:318:47: note: candidates are: ‘geometry_msgs::msg::PoseWithCovarianceStamped tf2::toMsg(const PoseWithCovarianceStamped&)’
  318 | geometry_msgs::msg::PoseWithCovarianceStamped toMsg(const geometry_msgs::msg::PoseWithCovarianceStamped& in)
      |                                               ^~~~~
/opt/ros/foxy/include/tf2_geometry_msgs/tf2_geometry_msgs.h:241:33: note:                 ‘geometry_msgs::msg::PoseStamped tf2::toMsg(const PoseStamped&)’
  241 | geometry_msgs::msg::PoseStamped toMsg(const geometry_msgs::msg::PoseStamped& in)
      |                                 ^~~~~
/opt/ros/foxy/include/tf2_geometry_msgs/tf2_geometry_msgs.h:174:34: note:                 ‘geometry_msgs::msg::PointStamped tf2::toMsg(const PointStamped&)’
  174 | geometry_msgs::msg::PointStamped toMsg(const geometry_msgs::msg::PointStamped& in)
      |                                  ^~~~~
/opt/ros/foxy/include/tf2_geometry_msgs/tf2_geometry_msgs.h:110:36: note:                 ‘geometry_msgs::msg::Vector3Stamped tf2::toMsg(const Vector3Stamped&)’
  110 | geometry_msgs::msg::Vector3Stamped toMsg(const geometry_msgs::msg::Vector3Stamped& in)
      |                                    ^~~~~
In file included from /opt/ros/foxy/include/tf2_geometry_msgs/tf2_geometry_msgs.h:35,
                 from /umd2_ws/src/umd_mission/umd_mission_core/plugins/actions/navigate_to_pose_action.cpp:19:
/opt/ros/foxy/include/tf2/convert.h:96:5: note:                 ‘template<class A, class B> B tf2::toMsg(const A&)’
   96 |   B toMsg(const A& a);
      |     ^~~~~
In file included from /umd2_ws/src/test/test_core/plugins/actions/navigate.cpp:19:
/opt/ros/foxy/include/tf2_geometry_msgs/tf2_geometry_msgs.h:364:76: error: ‘tf2::WithCovarianceStamped’ has not been declared
  364 | void fromMsg(const geometry_msgs::msg::PoseWithCovarianceStamped& in, tf2::WithCovarianceStamped<tf2::Transform>& out)
      |                                                                            ^~~~~~~~~~~~~~~~~~~~~
/opt/ros/foxy/include/tf2_geometry_msgs/tf2_geometry_msgs.h:364:97: error: expected ‘,’ or ‘...’ before ‘<’ token
  364 | void fromMsg(const geometry_msgs::msg::PoseWithCovarianceStamped& in, tf2::WithCovarianceStamped<tf2::Transform>& out)
      |                                                                                                 ^
/opt/ros/foxy/include/tf2_geometry_msgs/tf2_geometry_msgs.h:364:6: error: template-id ‘fromMsg<>’ for ‘void tf2::fromMsg(const PoseWithCovarianceStamped&, int)’ does not match any template declaration
  364 | void fromMsg(const geometry_msgs::msg::PoseWithCovarianceStamped& in, tf2::WithCovarianceStamped<tf2::Transform>& out)
      |      ^~~~~~~
/opt/ros/foxy/include/tf2_geometry_msgs/tf2_geometry_msgs.h:329:6: note: candidates are: ‘void tf2::fromMsg(const PoseWithCovarianceStamped&, geometry_msgs::msg::PoseWithCovarianceStamped&)’
  329 | void fromMsg(const geometry_msgs::msg::PoseWithCovarianceStamped& msg, geometry_msgs::msg::PoseWithCovarianceStamped& out)
      |      ^~~~~~~
/opt/ros/foxy/include/tf2_geometry_msgs/tf2_geometry_msgs.h:252:6: note:                 ‘void tf2::fromMsg(const PoseStamped&, geometry_msgs::msg::PoseStamped&)’
  252 | void fromMsg(const geometry_msgs::msg::PoseStamped& msg, geometry_msgs::msg::PoseStamped& out)
      |      ^~~~~~~
/opt/ros/foxy/include/tf2_geometry_msgs/tf2_geometry_msgs.h:185:6: note:                 ‘void tf2::fromMsg(const PointStamped&, geometry_msgs::msg::PointStamped&)’
  185 | void fromMsg(const geometry_msgs::msg::PointStamped& msg, geometry_msgs::msg::PointStamped& out)
      |      ^~~~~~~
/opt/ros/foxy/include/tf2_geometry_msgs/tf2_geometry_msgs.h:121:6: note:                 ‘void tf2::fromMsg(const Vector3Stamped&, geometry_msgs::msg::Vector3Stamped&)’
  121 | void fromMsg(const geometry_msgs::msg::Vector3Stamped& msg, geometry_msgs::msg::Vector3Stamped& out)
      |      ^~~~~~~
In file included from /opt/ros/foxy/include/tf2_geometry_msgs/tf2_geometry_msgs.h:35,
                 from /umd2_ws/src/umd_mission/umd_mission_core/plugins/actions/navigate_to_pose_action.cpp:19:
/opt/ros/foxy/include/tf2/convert.h:105:8: note:                 ‘template<class A, class B> void tf2::fromMsg(const A&, B&)’
  105 |   void fromMsg(const A&, B& b);
      |        ^~~~~~~
make[2]: *** [CMakeFiles/navigate_to_pose_action_bt_node.dir/build.make:63: CMakeFiles/navigate_to_pose_action_bt_node.dir/plugins/actions/navigate_to_pose_action.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:420: CMakeFiles/navigate_to_pose_action_bt_node.dir/all] Error 2
```
#### Additional information

When I use an older version of the ros-foxy-geometry-msgs `0.13.5-1focal.20201008.000653`. I don't see this. I think error started from this PR https://github.com/ros2/geometry2/pull/336
## Feature request

#### Feature description

#### Implementation considerations

clalancette, 2020-11-11:
For what it is worth, I can't reproduce this problem here with some packages that depend on `tf2_geometry_msgs`.  Can you give me a link to the code that you are trying to compile?

nnmm, 2020-11-13:
cc @JWhitleyWork 
I have been able to compile packages with dependencies on `tf2_geometry_msgs` as well.

ashwinsushil, 2020-11-15:
@clalancette @nnmm Pls give me a days time and I will create a docker with code that I'm trying to compile.

ashwinsushil, 2020-11-16:
> @clalancette @nnmm Pls give me a days time and I will create a docker with code that I'm trying to compile.

@clalancette @nnmm Please see the `Steps to reproduce issue` in the issue description to reproduce the error. In short, I created a docker image on which you can see this error.

clalancette, 2020-11-16:
> @clalancette @nnmm Please see the **Steps to reproduce issue** section in the issue description to reproduce the error. In short, I created a docker image in which you can see this error.

Did you also edit CMakeLists.txt to have:

```
find_package(tf2_geometry_msgs REQUIRED)

ament_target_dependencies(publisher_member_function rclcpp std_msgs tf2_geometry_msgs)
```

?

ashwinsushil, 2020-11-16:
> Did you also edit CMakeLists.txt to have:
>
> ```
> find_package(tf2_geometry_msgs REQUIRED)
>
> ament_target_dependencies(publisher_member_function rclcpp std_msgs tf2_geometry_msgs)
> ```
>
> ?

Just added! Updated the docker, but I get the same error.

[image]

nnmm, 2020-11-16:
I see that behavior in the docker container, but can't reproduce it outside with `minimal_publisher`. I see that there is an overlay workspace with `nav2_` packages in it, could that have anything to do with it?

clalancette, 2020-11-16:
> I see that behavior in the docker container, but can't reproduce it outside with `minimal_publisher`. I see that there is an overlay workspace with `nav2_` packages in it, could that have anything to do with it?

Ah, that's an interesting point.  Actually, now that I look at the error message more clearly, the problem seems to be that the version of tf2 is out-of-date one way or another.  @ashwinsushil What's the version of `ros-foxy-tf2` that you are using?  Also, does the nav overlay have tf2 in it?

nnmm, 2020-11-17:
@clalancette I think you got it right. I looked in the container with `apt list --installed`:

```
ros-foxy-tf2-geometry-msgs/focal,now 0.13.6-1focal.20201103.154904 amd64 [installed,automatic]
[...]
ros-foxy-tf2/now 0.13.5-1focal.20200911.214212 amd64 [installed,upgradable to: 0.13.6-1focal.20201028.172335]
```

nnmm, 2020-11-17:
@ashwinsushil `sudo apt upgrade` or equivalent should fix the problems you're seeing then.

ashwinsushil, 2020-11-17:
@clalancette @nnmm Just updated ros-foxy-tf! You guys are right `ros-foxy-tf2` was not updated. Nice catch! Sorry that I missed it.
Thanks a lot.

closed 2020-11-17 by ashwinsushil