ros2/geometry2 #346: `waitForTransform()` doesn't call callback when transform is already available

nnmm, 2020-11-13:
## Bug report

**Required Info:**

- Operating System: Ubuntu 20.04
- Installation type: Binaries
- Version or commit hash: Foxy
- DDS implementation: rmw_cyclonedds_cpp
- Client library (if applicable): rclcpp

#### Steps to reproduce issue
I made a simple [test package](https://github.com/nnmm/tf2_example) with one node that sends two transforms and one node that requests a transform that can be interpolated from those two. When the transforms are sent before the request, the callback is not called.
```
git clone git@github.com:nnmm/tf2_example.git
cd tf2_example
git checkout 5b837b82b15555f1cd5f35df3987b8e95c172e19 # for reproducibility in case the main branch changes
colcon build
source install/setup.bash
ros2 launch wait_for_transform wait_for_transform.launch.xml \
        executor:=single \
        method:=wft_callback \
        request_time:=current \
        send_first_tf:=before \
        send_second_tf:=before
```
The request looks like this: `tf_buffer_.waitForTransform("a", "b", request_time_, timeout_, callback_);` where `callback_` logs a message.

#### Expected behavior
The callback is executed and a line like
> [receiver-2] [INFO] [1605266908.513191150] [receiver.callback]: Calling get() ...

is logged

#### Actual behavior
The callback is not executed (based on the log message not appearing).

#### Additional information
This makes the callback useless if you don't know whether a transform has arrived yet. You could check with `canTransform` before calling `waitForTransform()`, but that means an obvious race condition.

ymd-stella, 2020-11-14:
A patch has been proposed to address this issue in https://github.com/orgs/ros2/projects/28.
Please check https://github.com/ros2/geometry2/pull/274.

jacobperron, 2020-11-16:
@clalancette or @ahcorde Could one of you take a look at backporting #274 to resolve this issue? Thanks :)

jacobperron, 2020-12-01:
@nnmm Are you able to confirm if https://github.com/ros2/geometry2/pull/353 fixes the issue for you on Foxy?

jacobperron, 2020-12-04:
Closing, assuming that https://github.com/ros2/geometry2/pull/353 fixed it. Please comment otherwise.

closed 2020-12-04 by jacobperron

nnmm, 2020-12-07:
@jacobperron Sorry for not replying in time – I attempted compiling the modified geometry2 code (in a docker container, which worked) and running the example from the bug report (which didn't work due to missing ros2 cli, and overlaying it onto a Foxy installation didn't work either). If it's still useful to you, I can try it out if you point me to some instructions.

jacobperron, 2020-12-07:
I was able to run your example in a docker containing (overlayed on Foxy). I had to uninstall the system version of geometry2 (which seemed to cause some Cmake collisions):

```
docker run -it ros:foxy
apt remove ros-foxy-tf2 ros-foxy-tf2-ros
mkdir -p ~/ws/src
cd ~/ws/src
git clone https://github.com/ros2/geometry2.git -b foxy
git clone https://github.com/nnmm/tf2_example.git
cd tf2_example
git checkout 5b837b82b15555f1cd5f35df3987b8e95c172e19
cd ~/ws/src
colcon build
source install/setup.bash
ros2 launch wait_for_transform wait_for_transform.launch.xml \
        executor:=single \
        method:=wft_callback \
        request_time:=current \
        send_first_tf:=before \
        send_second_tf:=before
```

FWIW, it looks like it may have been #281 (not #274) that fixed this issue.

nnmm, 2020-12-08:
Thank you, great @jacobperron!