ros2/geometry2 #350: Feature request: Port eigen_conversions to ROS 2

esteve, 2020-12-01:
## Feature request

#### Feature description
Port eigen_conversions from https://github.com/ros/geometry/tree/noetic-devel/eigen_conversions to ROS 2

#### Implementation considerations

clalancette, 2020-12-01:
This was actually done in #311 .  Does that work for you?

esteve, 2020-12-02:
@clalancette excellent, thanks!

closed 2020-12-02 by esteve

esteve, 2020-12-09:
@clalancette I realized that the `eigen_conventions` functions are not available in the `foxy` branch, but in `ros2`. Is there any chance these functions will backported to `foxy`? I can open a PR with the relevant changes.

clalancette, 2020-12-09:
> @clalancette I realized that the `eigen_conventions` functions are not available in the `foxy` branch, but in `ros2`. Is there any chance these functions will backported to `foxy`? I can open a PR with the relevant changes.

I don't see why not; it's a new package and all of the relevant things should be available in Foxy.  I see you've already opened a PR, so thanks!