ros2/geometry2 #354: Any plan for supporting sensor_msgs::msg::PointCloud2 transformation?

ldg810, 2020-12-04:
Hi, i am trying to use tf2::doTransform for sensor_msgs::msg::PointCloud2(which is the ros2 version msg)

I found that there is no available function for transform sensor_msgs::msg::PointCloud2. (Am I right?)

Thus, is there any plan to support ros2 system? (I am using dashing and foxy)

Thank you.

clalancette, 2020-12-04:
There is one in tf2_sensor_msgs.h: https://github.com/ros2/geometry2/blob/ad35d72a51137cc9caebe474c3e2b8aa0de0fff4/tf2_sensor_msgs/include/tf2_sensor_msgs/tf2_sensor_msgs.h#L61 .

Is that what you are looking for?  If not, can you describe in more detail what you are looking to do?

ldg810, 2020-12-10:
> There is one in tf2_sensor_msgs.h:
> 
> https://github.com/ros2/geometry2/blob/ad35d72a51137cc9caebe474c3e2b8aa0de0fff4/tf2_sensor_msgs/include/tf2_sensor_msgs/tf2_sensor_msgs.h#L61
> 
> .
> Is that what you are looking for? If not, can you describe in more detail what you are looking to do?

Thank you for reply! You are right. the doTransform for PointCloud2 was there.

I add tf2_sensor_msgs lib to CMakefile and include corresponding headerfile and succeed to use it.

closed 2020-12-10 by ldg810