ros2/geometry2 #357: Compare to none do not work with python 3.8

teyssieuman, 2020-12-14:
https://github.com/ros2/geometry2/blob/1574a6e6034d5af01b318558bde7fdee3bd3c692/tf2_ros/src/tf2_ros/buffer.py#L61

Since python 3.8, the above line should be written 

```python
if not cache_time is None:
```

clalancette, 2020-12-14:
> ```python
> if not cache_time is None:
> ```

Really it should be:

```python
if cache_time is not None:
```

Which is the preferred form nowadays, as I understand.  @teyssieuman would you mind opening a pull request with that change?

teyssieuman, 2020-12-15:
Works for me. The commit is updated.

clalancette, 2020-12-15:
Great.  Do you mind opening a pull request to the main repository here?  That way we can get this into Foxy (it is already fixed on master).

teyssieuman, 2020-12-15:
Do you think I should make the same pull request on the `ros2` branch ? or do I mistake ?
Maybe you want me to create a pull request from the `ros2` branch to the `foxy` branch ?

clalancette, 2020-12-15:
> Do you think I should make the same pull request on the `ros2` branch ? or do I mistake ?

No, that particular fix is already on the `ros2` branch.  So no need to fix it there.

> Maybe you want me to create a pull request from the `ros2` branch to the `foxy` branch ?

There is currently no pull request open against this repository for this problem.  So I'm suggesting that you open up a pull request from your fork to this repository, on the `foxy` branch.

teyssieuman, 2020-12-15:
Ok, what about this one :    https://github.com/teyssieuman/geometry2/pull/1

clalancette, 2020-12-15:
> Ok, what about this one : [teyssieuman#1](https://github.com/teyssieuman/geometry2/pull/1)

No, that is a pull request to your own repository.  You need to make one to this repository.

teyssieuman, 2020-12-15:
Ok, my mistake.

I still learn to use GitHub :innocent: 

Foxy #359 shall be better

clalancette, 2020-12-15:
> Foxy #359 shall be better

Yep!  That looks better.  Thanks.

PR #359 (2020-12-16): [Foxy]: Use 'is not None' instead of '!= None'

Fixes #357
closed 2020-12-27 by teyssieuman