ros2/geometry2 #363: On ros2:foxy, building geometry2 with "colcon build" failed

zhoulaifu, 2021-01-10:
Building geometry2 with "colcon build" produced the following error. Any idea why it happened? 

FYI,  I am on a docker machine of ros2 foxy. 

```
root@9da72d8e03df:~/ros2_ws# colcon build
Starting >>> tf2
Starting >>> tf2_msgs
Finished <<< tf2 [10.6s]
Starting >>> tf2_py
Starting >>> tf2_eigen_kdl
Finished <<< tf2_py [4.38s]
Finished <<< tf2_msgs [16.2s]
Starting >>> tf2_ros
Starting >>> tf2_ros_py
Finished <<< tf2_ros_py [0.80s]
Starting >>> tf2_tools
Starting >>> examples_tf2_py
Finished <<< tf2_tools [1.10s]
Finished <<< examples_tf2_py [1.15s]
Finished <<< tf2_eigen_kdl [14.7s]
--- stderr: tf2_ros
In file included from /root/ros2_ws/src/geometry2/tf2_ros/src/buffer.cpp:33:
/root/ros2_ws/src/geometry2/tf2_ros/include/tf2_ros/buffer.h:102:56: error: ‘from_nanoseconds’ is not a member of ‘rclcpp::Duration’
  102 |     const rclcpp::Duration timeout = rclcpp::Duration::from_nanoseconds(0)) const
      |                                                        ^~~~~~~~~~~~~~~~
/root/ros2_ws/src/geometry2/tf2_ros/include/tf2_ros/buffer.h:137:56: error: ‘from_nanoseconds’ is not a member of ‘rclcpp::Duration’
  137 |     const rclcpp::Duration timeout = rclcpp::Duration::from_nanoseconds(0)) const
      |                                                        ^~~~~~~~~~~~~~~~
/root/ros2_ws/src/geometry2/tf2_ros/include/tf2_ros/buffer.h:169:56: error: ‘from_nanoseconds’ is not a member of ‘rclcpp::Duration’
  169 |     const rclcpp::Duration timeout = rclcpp::Duration::from_nanoseconds(0),
      |                                                        ^~~~~~~~~~~~~~~~
/root/ros2_ws/src/geometry2/tf2_ros/include/tf2_ros/buffer.h:205:56: error: ‘from_nanoseconds’ is not a member of ‘rclcpp::Duration’
  205 |     const rclcpp::Duration timeout = rclcpp::Duration::from_nanoseconds(0),
      |                                                        ^~~~~~~~~~~~~~~~
make[2]: *** [CMakeFiles/tf2_ros.dir/build.make:63: CMakeFiles/tf2_ros.dir/src/buffer.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:493: CMakeFiles/tf2_ros.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< tf2_ros [13.3s, exited with code 2]

Summary: 7 packages finished [29.8s]
  1 package failed: tf2_ros
  1 package had stderr output: tf2_ros
  7 packages not processed
```

wasowski, 2021-01-10:
Are you sure you have run `rosdep install --from-paths src -y` ?

zhoulaifu, 2021-01-10:
Thanks Andrzej. I did not, unaware of it. But when I do what you said, I got some additional errors:

```
root@9da72d8e03df:~/ros2_ws# rosdep install --from-paths src -y
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
examples_tf2_py: Cannot locate rosdep definition for [tf2_ros_py]
tf2_sensor_msgs: Cannot locate rosdep definition for [tf2_ros_py]
tf2_geometry_msgs: Cannot locate rosdep definition for [tf2_ros_py]
test_tf2: Cannot locate rosdep definition for [tf2_ros_py]
tf2_tools: Cannot locate rosdep definition for [tf2_ros_py]
geometry2: Cannot locate rosdep definition for [tf2_eigen_kdl]
tf2_kdl: Cannot locate rosdep definition for [tf2_ros_py]
```

zhoulaifu, 2021-01-10:
I just solved this problem. It is due to my misuse. I should have used 

`git checkout $ROS_DISTRO`

after cloning the repository.

closed 2021-01-10 by zhoulaifu