ros2/geometry2 #365: Unit tests for tf2_sensor_msgs.cpp not executed?

zhoulaifu, 2021-01-13:
It seems that the program `geometry2/tf2_sensor_msgs/test/test_tf2_sensor_msgs.cpp` is not executed when running  `colcon test --packages-select tf2_sensor_msgs`,  as discussed in this thread: https://answers.ros.org/question/369345/where-is-the-binary-of-test_tf2_sensor_msgs/, which refers to your CMakefLists https://github.com/ros2/geometry2/blob/99d87c508ba16743837910381775849110ffe52f/tf2_sensor_msgs/CMakeLists.txt#L28-L45

clalancette, 2021-01-13:
That's right, we've never ported the tests from ROS 1 to ROS 2.  Any help in doing it would be appreciated!

zhoulaifu, 2021-01-13:
What would be the challenge to run or incorporate the tests?  I tried to run the generated binary in build/tf2_geometry_msgs/test_tf2_geometry_msgs manually, and the tests passed well. See the screenshot below. 
 
[image]

clalancette, 2021-01-13:
> What would be the challenge to run or incorporate the tests? I tried to run the generated binary in build/tf2_geometry_msgs/test_tf2_geometry_msgs manually, and the tests passed well. See the screenshot below.

That's a different test in a different package.  The one you linked to above was `test_tf2_sensor_msgs_cpp` from the `tf2_sensor_msgs` package.  To enable the tests in `tf2_sensor_msgs`, you'd have to port `test_tf2_sensor_msgs.cpp` and `test_tf2_sensor_msgs.py` to ROS 2 (which both look straightforward), and then integrate them into the CMakeLists.txt.  There are some examples of doing similar things in https://github.com/ros2/geometry2/blob/ros2/tf2/test/simple_tf2_core.cpp , for example.

(incidentally, please copy-n-paste output rather than taking screenshots; screenshots aren't searchable)

zhoulaifu, 2021-01-14:
I uncommented some testing part in CMakeLists.txt in tf2_sensor_msgs and ran `colcon build`. The compilation passed well but the expected binnary `test_tf2_sensor_msgs_cpp` is not generated. Any idea what happened? Below is the modified CMakeLists.txt of the tf2_sensor_msgs package.

```
cmake_minimum_required(VERSION 3.5)
project(tf2_sensor_msgs)

# Default to C++14
if(NOT CMAKE_CXX_STANDARD)
  set(CMAKE_CXX_STANDARD 14)
endif()
if(CMAKE_COMPILER_IS_GNUCXX OR CMAKE_CXX_COMPILER_ID MATCHES "Clang")
  add_compile_options(-Wall -Wextra)
endif()

find_package(ament_cmake_auto REQUIRED)
set(required_dependencies
  "sensor_msgs"
  #"python_orocos_kdl"
  "tf2"
  "tf2_ros"
)

ament_auto_find_build_dependencies(REQUIRED ${required_dependencies})

find_package(eigen3_cmake_module REQUIRED)
find_package(Eigen3 REQUIRED)

ament_export_dependencies(eigen3_cmake_module)
ament_export_dependencies(Eigen3)

# TODO enable tests
#if(BUILD_TESTING)
#  catkin_add_nosetests(test/test_tf2_sensor_msgs.py)

find_package(ament_cmake REQUIRED COMPONENTS
  sensor_msgs
  rostest
  tf2_ros
  tf2
)
include_directories(${EIGEN_INCLUDE_DIRS})
add_executable(test_tf2_sensor_msgs_cpp EXCLUDE_FROM_ALL test/test_tf2_sensor_msgs.cpp)
target_link_libraries(test_tf2_sensor_msgs_cpp ${catkin_LIBRARIES} ${GTEST_LIBRARIES})
#if(TARGET tests)
#  add_dependencies(tests test_tf2_sensor_msgs_cpp)
#endif()
#add_rostest(${CMAKE_CURRENT_SOURCE_DIR}/test/test.launch)
#endif()

ament_auto_package()

```

clalancette, 2021-01-14:
> ```
> # TODO enable tests
> #if(BUILD_TESTING)
> #  catkin_add_nosetests(test/test_tf2_sensor_msgs.py)
>
>
> find_package(ament_cmake REQUIRED COMPONENTS
>   sensor_msgs
>   rostest
>   tf2_ros
>   tf2
> ) [...]

This whole section should be more like:

```
if(BUILD_TESTING)
  find_package(ament_cmake_gtest REQUIRED)

  ament_add_gtest(test_tf2_sensor_msgs_cpp test/test_tf2_sensor_msgs.cpp)
  target_link_libraries(test_tf2_sensor_msgs_cpp ${PROJECT_NAME})
endif()

ament_auto_package()
```

You may also need to add some additional `target_include_directories` for EIGEN; I'm not sure.  With something like that, it should be able to run under `colcon test`.

CursedRock17, 2024-03-25:
Did PR #422 fully resolve this, or can I pick this up?

clalancette, 2024-03-25:
> Did PR #422 fully resolve this, or can I pick this up?

Yeah, it looks like it was done, so I'll close this out.

closed 2024-03-25 by clalancette