ros2/geometry2 #366: Message Filter stuck dropping stale messages due to large queue size

Jconn, 2021-01-14:
## Bug report

**Required Info:**

- Operating System:
Ubuntu 18.04
- Installation type:
Source
- Version or commit hash:
  - 
- DDS implementation:
cyclone
- Client library (if applicable):
rclcpp

Very rarely, my system will enter an error state where all of my message filters start rejecting my messages.
My log output ends up looking like this until I manually kill my nodes:

```
[1610551896.439708854] [local_costmap.local_costmap_rclcpp_node]: Message Filter dropping message: frame 'camera_link' at time 1610551894.931 for reason 'Unknown'
[1610551896.472888675] [local_costmap.local_costmap_rclcpp_node]: Message Filter dropping message: frame 'camera_link' at time 1610551894.960 for reason 'Unknown'
[1610551896.505286865] [local_costmap.local_costmap_rclcpp_node]: Message Filter dropping message: frame 'camera_link' at time 1610551894.990 for reason 'Unknown'
[1610551896.536371604] [local_costmap.local_costmap_rclcpp_node]: Message Filter dropping message: frame 'camera_link' at time 1610551895.020 for reason 'Unknown'
[1610551896.560983603] [local_costmap.local_costmap_rclcpp_node]: Message Filter dropping message: frame 'camera_link' at time 1610551895.050 for reason 'Unknown'
[1610551896.587268162] [local_costmap.local_costmap_rclcpp_node]: Message Filter dropping message: frame 'camera_link' at time 1610551895.080 for reason 'Unknown'
[1610551896.619227590] [local_costmap.local_costmap_rclcpp_node]: Message Filter dropping message: frame 'camera_link' at time 1610551895.110 for reason 'Unknown'
[1610551896.654364589] [local_costmap.local_costmap_rclcpp_node]: Message Filter dropping message: frame 'camera_link' at time 1610551895.140 for reason 'Unknown'
[1610551896.677907253] [local_costmap.local_costmap_rclcpp_node]: Message Filter dropping message: frame 'camera_link' at time 1610551895.170 for reason 'Unknown'
[1610551896.704027512] [local_costmap.local_costmap_rclcpp_node]: Message Filter dropping message: frame 'camera_link' at time 1610551895.201 for reason 'Unknown'
[1610551896.742694255] [local_costmap.local_costmap_rclcpp_node]: Message Filter dropping message: frame 'camera_link' at time 1610551895.230 for reason 'Unknown'
[1610551896.762921561] [local_costmap.local_costmap_rclcpp_node]: Message Filter dropping message: frame 'camera_link' at time 1610551895.260 for reason 'Unknown'
[1610551896.794901727] [local_costmap.local_costmap_rclcpp_node]: Message Filter dropping message: frame 'camera_link' at time 1610551895.290 for reason 'Unknown'
[1610551896.820790032] [local_costmap.local_costmap_rclcpp_node]: Message Filter dropping message: frame 'camera_link' at time 1610551895.320 for reason 'Unknown'
[1610551896.855324540] [local_costmap.local_costmap_rclcpp_node]: Message Filter dropping message: frame 'camera_link' at time 1610551895.350 for reason 'Unknown'
[1610551896.889219195] [local_costmap.local_costmap_rclcpp_node]: Message Filter dropping message: frame 'camera_link' at time 1610551895.383 for reason 'Unknown'
[1610551896.914657912] [local_costmap.local_costmap_rclcpp_node]: Message Filter dropping message: frame 'camera_link' at time 1610551895.410 for reason 'Unknown'

```
I've confirmed that my transforms are still being published. 

Taking a look at the implementation, my problem seems to be at the [transform check](https://github.com/ros2/geometry2/blob/ros2/tf2_ros/include/tf2_ros/message_filter.h#L556). Something causes my message filter queue to fill up. My queue depth is so large that the last message in the queue is too old to find a valid transform. Because only one message in the queue is processed per transform, and the sensor topic publishes at the same rate as my transforms, the oldest message in my queue continues to be stale. 

I'm wondering why the Message Filter only processes one message per transform? My thinking is that if as many messages as possible were processed on every transform arrival, then I could avoid this stale data at the end of my queue permanently disabling operation.

BriceRenaudeau, 2021-06-30:
Hello, I may have a similar issue. [Ubuntu 20.04 - ROS Galactic - CycloneDDS - Geometry2 from release]
On normal conditions, the Nav2 `controller_server` node works fine using 20% CPU.
Then for no apparent reasons, we get the error message:
>[controller_server][ERROR][local_costmap.local_costmap]: TF Exception for sensor frame: , cloud frame: camera_right_depth_optical_frame, lookup would require extrapolation onti the future.

After that, the Message Filter is stuck dropping message and the CPU consumption of the node gets over 120%
>controller_server][INFO][local_costmap.local_costmap]: Message Filter dropping message: frame 'laser' at time ... for reason 'discarding message because the queue is full'

The only way to recover is to deactivate and reactivate the lifecycle nodes.

The flame graph when the issue happens looks like this.
[image]
We can see that more than 60% of the execution time is taken by the tf2_ros functions.

doisyg, 2021-08-11:
> Very rarely, my system will enter an error state where all of my message filters start rejecting my messages.

Same for us (same system as @BriceRenaudeau) , it happens very rarely, once a week on a constantly running system.

@Jconn Did you manage to find the cause ?

I wonder of the issue come from the controller_server (https://github.com/ros-planning/navigation2/issues/2333 ) or tf

Jconn, 2021-08-11:
We never solved this. This stuck state seemed to happen to us more often when our cpu load was higher e.g. We would hit this case more often with rviz running.

doisyg, 2021-08-11:
All right, I managed to reproduce it more consistently and more frequently. I will report here

doisyg, 2021-08-18:
Solved for obstacle_layer and STVL by using a timeout on the message filter, see here: https://github.com/ros-planning/navigation2/issues/2333
However I still believe there is a bug in message filter or a major performance bottleneck arising from the moment there is an extrapolation error.
I believe this causes a similar issue in rviz