ros2/geometry2 #370: state_estimation_publisher does not work reliably in Foxy

nnmm, 2021-01-29:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - Just the `ros:foxy` Docker image
- Version or commit hash:
  - tf2 libraries are at version 0.13.9
- DDS implementation:
  - I don't know, but it's the default one in that container
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue

```
$ docker run -it --rm ros:foxy
docker$ apt update && apt upgrade && apt install -y tmux
docker$ tmux
docker$ tmux split-pane -h
```

In the first panel:
```
docker$ ros2 topic echo /tf_static tf2_msgs/msg/TFMessage
```
In the other panel:
```
docker$ ros2 run tf2_ros static_transform_publisher -57.60810852050781 -41.382755279541016 0 0 0 0 map odom
```
Now, repeatedly shutdown and restart the publisher and observe how many messages you see in the first panel (that might require e.g. hitting Enter in the first panel once it's full to see when new messages are arriving). 

#### Expected behavior
A message is echoed each time the static_transform_publisher is started.

#### Actual behavior
About 1 in 4 times, no message is echoed. The rate might be different depending on the configuration, I had much lower flakiness at some other point.

#### Additional information
`--qos-durability=transient_local` also doesn't help.

clalancette, 2021-01-30:
If you add in `--qos-reliability=reliable`, does that help?

nnmm, 2021-01-30:
@clalancette Yes, thanks a lot! Is that expected behavior then?

clalancette, 2021-01-30:
> @clalancette Yes, thanks a lot! Is that expected behavior then?

Yes and no.  The proper QoS for the `/tf_static` topic is indeed Transient Local/Reliable, so from that perspective the subscriber has to match.  However, it would be much nicer if `ros2 topic echo` could automatically figure that out.  I remember discussing this in an issue at some point, but I can't find it at the moment.  So my suggestion is to close this issue (as there isn't something particularly wrong with geometry2), and instead open a new issue over at https://github.com/ros2/ros2cli requesting that `ros2 topic echo` be able to figure this out.

nnmm, 2021-02-01:
Okay, I opened one (well, actually two because I missed that QoS mismatch warnings are already implemented) here: https://github.com/ros2/ros2cli/issues/594

closed 2021-02-01 by nnmm