ros2/geometry2 #372: PoseWithCovarianceStamped transform doesn't transform covariance part

ksuszka, 2021-02-02:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04, but it doesn't really matter
- Installation type:
  - source code analysis
- Version or commit hash:
  - a4f152f67d09b50d6d6450bb00c44512e52a880c
- DDS implementation:
  - N/A
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

```

```

#### Expected behavior
doTransform on PoseWithCovarianceStamped should also transform covariance.
#### Actual behavior
doTransform on PoseWithCovarianceStamped copies covariance without transforming it.

#### Additional information
The culprit seems to be line https://github.com/ros2/geometry2/blob/d65b7eb996dcc81bfd3389153b25b9d56eed0e8c/tf2_geometry_msgs/include/tf2_geometry_msgs/tf2_geometry_msgs.h#L309

It seems that this functionality is implemented in ROS1 version https://github.com/ros/geometry2/blob/c73b5939723db078c9bbe18523230ad54f859682/tf2_geometry_msgs/include/tf2_geometry_msgs/tf2_geometry_msgs.h#L927

clalancette, 2021-02-02:
Yep, agreed.  A pull request to port that functionality over from the ROS 1 version would be welcome.

aprotyas, 2021-06-07:
On that note, `do_transform_pose_with_covariance_stamped` in `tf2_geometry_msgs.py` does not transform the covariance either. Same pattern.
https://github.com/ros2/geometry2/blob/6fa2001c026ff104b890663b52f2e3d518140146/tf2_geometry_msgs/src/tf2_geometry_msgs/tf2_geometry_msgs.py#L143

Since I'm working on this issue, should I fold above changes into the same PR? @clalancette

clalancette, 2021-06-07:
> Since I'm working on this issue, should I fold above changes into the same PR? @clalancette

It's worthwhile to fix it both places, but I'll suggest two separate PRs, one for C++ and one for Python.

closed 2021-06-11 by aprotyas

msmcconnell, 2021-12-21:
@clalancette Is there any chance the fix for this in #430 gets backported to Foxy?

clalancette, 2021-12-21:
> @clalancette Is there any chance the fix for this in #430 gets backported to Foxy?

I think that is viable; it doesn't change API or ABI.  Mind opening a backport PR?

aprotyas, 2021-12-21:
Opened 

> > @clalancette Is there any chance the fix for this in #430 gets backported to Foxy?
> 
> I think that is viable; it doesn't change API or ABI. Mind opening a backport PR?

I've opened #488 and #489 through mergify.

msmcconnell, 2021-12-21:
Thanks for putting those up @aprotyas ! Hope they can make it in soon.

aprotyas, 2021-12-23:
FYI: both backports have landed.

msmcconnell, 2021-12-27:
@aprotyas Awesome thanks for doing that!

PR #430 (2021-06-11): `tf2_geometry_msgs`: Fixing covariance transformation in `doTransform<PoseWithCovarianceStamped, TransformStamped>`
Previously, the covariance matrix of a `PoseWithCovarianceStamped` message [was incorrectly transformed](https://github.com/ros2/geometry2/blob/d65b7eb996dcc81bfd3389153b25b9d56eed0e8c/tf2_geometry_msgs/include/tf2_geometry_msgs/tf2_geometry_msgs.h#L309). This PR:

- Adds a `transformCovariance` method to address the covariance transformation.
- Updates the `TfGeometry/FrameWithCovariance` unit test to check for the expected transformed covariance.

Fixes #372
commit ae30f0ba (2021-06-11): `tf2_geometry_msgs`: Fixing covariance transformation in `doTransform<PoseWithCovarianceStamped, TransformStamped>` (#430)

Fixes #372.

* Implement covariance transformation (cpp).

* Fixed `TfGeometry/FrameWithCovariance` test. The updated test checks against the correctly transformed
covariance matrix.
commit a28ae8bc (2021-12-21): `tf2_geometry_msgs`: Fixing covariance transformation in `doTransform<PoseWithCovarianceStamped, TransformStamped>` (#430)

Fixes #372.

* Implement covariance transformation (cpp).

* Fixed `TfGeometry/FrameWithCovariance` test. The updated test checks against the correctly transformed
covariance matrix.

(cherry picked from commit ae30f0ba0cea64a13347a90ce4945e4452d7cf37)
commit f84d636f (2021-12-21): `tf2_geometry_msgs`: Fixing covariance transformation in `doTransform<PoseWithCovarianceStamped, TransformStamped>` (#430)

Fixes #372.

* Implement covariance transformation (cpp).

* Fixed `TfGeometry/FrameWithCovariance` test. The updated test checks against the correctly transformed
covariance matrix.

(cherry picked from commit ae30f0ba0cea64a13347a90ce4945e4452d7cf37)