ros2/geometry2 #373: [tf2_echo] Exception thrown:"X" passed to lookupTransform argument target_frame does not exist

jacobperron, 2021-02-04:
## Bug report

**Required Info:**

- Operating System:
  - OSX (High Sierra)
- Installation type:
  - Binary
- Version or commit hash:
  - Crystal (pre-release)
- DDS implementation:
  - Fast-RTPS, Connext (5.3.1), and Opensplice (6.9)

#### Steps to reproduce issue

Follow the [tf2 tutorial](https://index.ros.org/doc/ros2/tf2/).
In one terminal:
```
ros2 run tf2_ros static_transform_publisher 1 2 3 0.5 0.1 -1.0 foo bar
```

In a second terminal:
```
ros2 run tf2_ros tf2_echo foo bar
```

#### Expected behavior

Second terminal (`tf2_echo`) starts outputting:

```
At time 0.0
- Translation: [1.000, 2.000, 3.000]
- Rotation: in Quaternion [-0.475, -0.076, 0.240, 0.843]
```

#### Actual behavior

`tf2_echo` outputs the following error message one or more times before we see the expected output:
```
Failure at 1.54475e+09
Exception thrown:"foo" passed to lookupTransform argument target_frame does not exist. 
The current list of frames is:

```

#### Additional information

The time between the exception error message and the expected output is especially long if the transform publisher is using Fast-RTPS and `tf2_echo` is using OpenSplice.

clalancette, 2021-02-04:
As we discussed, it would be nice if the `tf2_echo` tool didn't print out this error at the beginning.  On the other hand, if there is no component in the system that *ever* publishes that particular transform, we should inform the user of that.

One way to improve on this is to wait longer in the `tf2_echo` program before printing an error.  That is, try to get the transform for X seconds at the beginning, and only print the error if we can't get it.

The other way to improve things is to throttle this message so it only prints out every so often.

jacobperron, 2021-02-10:
I just took a look at this again. Looks like the issue was resolved in https://github.com/ros2/geometry2/pull/254 by adding a loop to wait for the requested transform to be ready.

closed 2021-02-10 by jacobperron