ros2/geometry2 #377: The test test_message_filter of test_tf2 fails sometimes, but not always

zhoulaifu, 2021-02-08:
On ROS foxy, sometimes (not always) `test_tf2/test/test_message_filter` fails. 

```
/opt/ros_ws/src/src/geometry2/test_tf2/test/test_message_filter.cpp:217: Failure
Expected equality of these values:
  messages
    Which is: 30
  n.count_
    Which is: 29
[  FAILED  ] MessageFilter.concurrentTransforms (658 ms)
[ RUN      ] MessageFilter.setTargetFrame
[       OK ] MessageFilter.setTargetFrame (11 ms)
[ RUN      ] MessageFilter.multipleTargetFrames
[       OK ] MessageFilter.multipleTargetFrames (8 ms)
[ RUN      ] MessageFilter.tolerance
[       OK ] MessageFilter.tolerance (11 ms)
[----------] 8 tests from MessageFilter (751 ms total)

[----------] Global test environment tear-down
[==========] 8 tests from 1 test case ran. (751 ms total)
[  PASSED  ] 7 tests.
[  FAILED  ] 1 test, listed below:
[  FAILED  ] MessageFilter.concurrentTransforms

 1 FAILED TEST 
```
I used this command line of a ros:foxy docker machine:

`for i in {1..50};  do ./build/test_tf2/test_message_filter; done`

clalancette, 2021-02-08:
I'm not able to reproduce this on the latest, so there is probably something we need to backport to Foxy.

zhoulaifu, 2021-02-08:
Thanks Chris!

Just want to make sure: does "docker pull ros:rolling" gives us the latest version of ROS? 

And for geometry2, it is "ros2" tag that corresponds to your latest version?

clalancette, 2021-02-08:
> Just want to make sure: does "docker pull ros:rolling" gives us the latest version of ROS?

The way that this works is that we do all of the development on the "master" or "ros2" branches of our repositories.  Periodically (every 2 - 4 weeks), we do releases of that code into Rolling.  So Rolling isn't always the latest, it will lag behind the source versions a bit.

> And for geometry2, it is "ros2" tag that corresponds to your latest version?

Yes, for this repository, the "ros2" branch is the latest.

clalancette, 2024-12-04:
Given that this is all about Foxy and Galactic, both of which are End-of-Life, I'm going to close this out.  If this is still a problem on the currently supported versions (Humble or Jazzy), please feel free to file another issue.

PR #449 (2021-08-31): 👩‍🌾[backport] Fix a TOCTTOU race in tf2. (#307)
Backport #307 with the goal of fixing a flaky test on Foxy (closes #377).

See more context on #377.

---

https://github.com/osrf/buildfarmer/issues/224
closed 2024-12-04 by clalancette