ros2/geometry2 #379: Porting tf2_geometry_msgs

karl-schulz, 2021-02-16:
Transforming geometry and sensor messages in Python is currently not possible, as tf2_geometry_msgs is excluded from the build. 

However, I was able to get it running fine, by installing PyKDM with these files:
* https://github.com/ros2/geometry2/blob/foxy/tf2_sensor_msgs/src/tf2_sensor_msgs/tf2_sensor_msgs.py 
* https://github.com/ros2/common_interfaces/blob/master/sensor_msgs_py/sensor_msgs_py/point_cloud2.py

Is there a reason to still exclude tf2_geometry_msgs and tf2_sensor_msgs? Using them in python is quite useful (although slow).

Thanks!

clalancette, 2021-02-16:
See https://github.com/ros2/geometry2/pull/360 , where we are (slowly) trying to get this to work.

karl-schulz, 2021-02-16:
Oh, thanks - I was using the search but somehow didn't find this one. Thanks alot!

closed 2021-02-16 by karl-schulz