ros2/geometry2 #387: Tutorials: Write a tf2 intermediate tutorial

clalancette, 2021-03-04:
One of the very important pieces of ROS 2 is the TF2 library.  Understanding how TF2 works and how to use the library is a critical skill for ROS 2 users.

To this end, we should have a tutorial that discusses TF2 in some detail and how to use it.  We have the following rough outline for it so far:

* Needs a much more in depth concept overview before code tutorials start 
* Tutorials
    * Introduction to tf
    * Writing a tf broadcaster (C++/Python)
    * Writing a tf listener (C++/Python)
    * Adding a frame (C++/Python)
    * Learning about tf and time (C++/Python)
    * Time travel with tf (C++/Python)
    * Debugging tf problems
    * Using Stamped datatypes with tf::MessageFilter
    * Setting up your robot using tf
    * Using the robot state publisher on your own robot
    * Using urdf with robot_state_publisher
    *Conclusion?
* Process
    * Port from wiki
* Some points to cover
    * Probably all included in the original tutorials
    * Anything changed significantly from tf -> tf2
    * Emphasize “what” and “why” in each tutorial, make sure none of them just start without context (they’re kind of like that now)
* Resources
    * https://index.ros.org/doc/ros2/Tutorials/tf2/
    * http://wiki.ros.org/tf/Tutorials
    * http://wiki.ros.org/tf - expand on this for conceptual overview

kurshakuz, 2021-06-01:
Hello @clalancette!

I am currently working on this issue and porting tf2 tutorials and would like to know if you could suggest the best way to update the tutorials structure. I mentioned all details in the following issue: https://github.com/ros2/ros2_documentation/issues/1631

clalancette, 2021-09-24:
I'm actually going to close this issue, in favor of https://github.com/ros2/ros2_documentation/issues/1631

closed 2021-09-24 by clalancette