ros2/geometry2 #389: Node namespace is used for TransformListener topics in tf2_ros_py

jacobperron, 2021-03-09:
## Bug report

**Required Info:**

- Operating System:
  - All
- Installation type:
  - Any
- Version or commit hash:
  - latest
- DDS implementation:
  - N/A
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue

Create a Python TransformListener and run it with a node namespace.

```py
# tf_listen.py
import sys
import rclpy
from tf2_ros.buffer import Buffer
from tf2_ros.transform_listener import TransformListener

if __name__ == '__main__':
    rclpy.init(args=sys.argv)
    node = rclpy.create_node('foo')
    buf = Buffer()
    listener = TransformListener(buf, node)
    rclpy.spin(node)
    rclpy.shutdown()
```

```sh
python3 tf_listen.py --ros-args -r __ns:=/bar
```

Then, check the topics:

```sh
ros2 topic list
```

#### Expected behavior

We expect the listener to subscribe to `/tf` and `/tf_static` topics.

#### Actual behavior

The listener to subscribe to `/bar/tf` and `/bar/tf_static` topics.

#### Additional information

This behvaviour is inconsistent with the TransformBroadcaster and StaticTransformBroadcaster, which publisher to the global namespace `/tf` and `/tf_static` respectively.

Also, this behaviour is inconsistent with the C++ implementation of the transform listener **and** the[ ROS 1 Python implementation](https://github.com/ros/geometry2/blob/c73b5939723db078c9bbe18523230ad54f859682/tf2_ros/src/tf2_ros/transform_listener.py#L56-L57).

Past tickets related to using a global namespace for the `/tf` and `/tf_static` topics: 

- https://github.com/ros2/geometry2/pull/55
- https://github.com/ros2/geometry2/issues/54
- https://github.com/ros2/geometry2/pull/326

jacobperron, 2021-03-09:
Proposed fix: https://github.com/ros2/geometry2/pull/390

closed 2021-03-10 by clalancette

apockill, 2022-02-05:
Oh man, I just spent two days debugging this. 

I'm relatively new to ROS, and I was curious- how do people typically get access to upstream fixes? For example, I'm on `foxy` currently. Would I need to wait for the next ROS2 LTS release to benefit from these changes?

clalancette, 2022-02-07:
> I'm relatively new to ROS, and I was curious- how do people typically get access to upstream fixes? For example, I'm on `foxy` currently. Would I need to wait for the next ROS2 LTS release to benefit from these changes?

In short: it depends.  Primary development on ROS 2 happens on the Rolling branches, which feed into the next release.  So all of the fixes are guaranteed to be in the newest release, but of course there is a year gap between releases.  Since this release went in in March 2021, that means the fix is available in Galactic and Rolling (and will of course be available in Humble when we branch off for that in a couple of months).

We also do backports to older, supported distributions, but on a case-by-case basis.  We have a very strict API/ABI compatibility policy for already released distributions, which is why we don't do this across the board.

This particular case is tricky.  The fix in #390 is the correct thing to do, but if we change it in Foxy, we could be breaking users who are depending on this behavior.  Nevertheless, if you would like to open a backport PR, we can consider it further there.

apockill, 2022-02-08:
Thank you for the response! I understand not wanting to break the API/ABI compatibility for already released distributions- especially for this fix. 

For anyone else coming to this thread looking for a workaround, I simply remapped `/namespace/tf_static` to `/tf_static` and did the same for `/tf`.

delihus, 2024-01-11:
Hi,
what about using TransformListener with namespace?
The nav2 multirobots demos uses separate `tf` topic for every robot:
https://github.com/ros-planning/navigation2/blob/main/nav2_bringup/launch/bringup_launch.py#L58

I propose the `use_global_namespace` parameter.

PR #390 (2021-03-10): Use global namespace for TransformListener topics
Adding a forward slash to the topic names ensures that when the associated node is namespaced, the topics are not.
This is the same behavior as the C++ counterpart and the ROS 1 implementation.

Fixes #389
commit 2737edbc (2021-03-10): Use global namespace for TransformListener topics (#390)

Adding a forward slash to the topic names ensures that when the associated node is namespaced, the topics are not.
This is the same behavior as the C++ counterpart and the ROS 1 implementation.

Fixes #389