ros2/geometry2 #405: canTransform does not suppress warning when called with error_msg

msy22, 2021-04-28:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - ROS2 Foxy, installed in Docker container `osrf/ros:foxy-desktop` from [hub.docker.com](https://hub.docker.com/r/osrf/ros/)
- Version or commit hash:
  - As per latest version of `osrf/ros:foxy-desktop` from [hub.docker.com](https://hub.docker.com/r/osrf/ros/)

I'm trying to find a work-around for the issue described in several Github issues, and addressed in [issue #396](https://github.com/ros2/geometry2/issues/396), where calling canTransform directly or indirectly through lookupTransform results in an error message being spammed to the terminal.

In the process of looking for a way to suppress this output, I found [this comment by tfoote in issue #358](https://github.com/ros2/geometry2/issues/358#issuecomment-772957229). The error is generated by function [fillOrWarnMessageForInvalidFrame](https://github.com/ros2/geometry2/blob/a1be44fdbd8c032351ffd3ddff8467d000f4c6d7/tf2/src/buffer_core.cpp#L145) in buffer_core.cpp, and tfoote's comment suggests that calling canTransform with the error_msg parameter != nullptr will result in the function filling the error_msg string instead of generating the error message (i.e. the error message is no longer spammed to terminal and is instead handed to the caller to deal with). And if we look at the code for this:

```
if (error_msg != nullptr) {
  *error_msg = s;
} else {
  CONSOLE_BRIDGE_logWarn("%s", s.c_str());
}
```

unless I am missing something, this does seem to be how this code should work. But it doesn't. I still see the error (see below)

#### Steps to reproduce issue

```
std::string suppressed_error = "dummy_str";
if (buffer->canTransform(
    target_frame,
    source_frame,
    rclcpp::Time(0),
    frame_timeout,
    &suppressed_error))
{
    std::cout<< "Found " << target_frame << " -> " << source_frame << " transform" << std::endl;
}
```

#### Expected behavior

I expect the call to [CONSOLE_BRIDGE_logWarn](https://github.com/ros2/geometry2/blob/a1be44fdbd8c032351ffd3ddff8467d000f4c6d7/tf2/src/buffer_core.cpp#L156) to not be printed because the error_msg is non-null. So I expect to just see this:

```
[tf_broadcaster_exe-1] Found base_link -> utm transform
```

#### Actual behavior

Instead I see the same error message being spammed until the transform is found:

```
[tf_broadcaster_exe-1] Warning: Invalid frame ID "utm" passed to canTransform argument target_frame - frame does not exist
[tf_broadcaster_exe-1]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.9/src/buffer_core.cpp
[tf_broadcaster_exe-1] Warning: Invalid frame ID "utm" passed to canTransform argument target_frame - frame does not exist
[tf_broadcaster_exe-1]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.9/src/buffer_core.cpp
[tf_broadcaster_exe-1] Warning: Invalid frame ID "utm" passed to canTransform argument target_frame - frame does not exist
[tf_broadcaster_exe-1]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.9/src/buffer_core.cpp
[tf_broadcaster_exe-1] Warning: Invalid frame ID "utm" passed to canTransform argument target_frame - frame does not exist
[tf_broadcaster_exe-1]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.9/src/buffer_core.cpp
[tf_broadcaster_exe-1] Found base_link -> utm transform
```

I'm sure this is a bug, but I haven't yet identified the cause.

jlack1987, 2022-06-22:
I can report this is still an issue. A very annoying issue at that!

jlack1987, 2022-07-21:
It looks like in the ros1 version of this stuff this was fixed [here](https://github.com/ros/geometry2/pull/469). If we could get this same fix here :pray: