ros2/geometry2 #407: tf2_echo does not work with --ros-args

PGotzmann, 2021-04-29:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - binaries
- Version or commit hash:
  - 0.13.10 
- DDS implementation:
  - CycloneDDS
- Client library (if applicable):
  - N/A 

#### Steps to reproduce issue
Run `tf2_echo` with an additional `ros-args` parameter e.g.:
```
ros2 run tf2_ros tf_echo --ros-args -r /tf:=/bar/tf
```

#### Expected behavior
`ros-args` parameters such as remappings can be used with `tf2_echo` (as it is possible with tf2_monitor, buffer_server, static_transform_publisher).

#### Actual behavior
If additional `ros-args` parameters are provided  only the usage information is shown and program is terminated.

#### Additional information
I have had a quick look at the source code and it looks like the reason for that is that the ros-args are not removed from command line arguments here, thus the check on the number of expected arguments in [f2_echo.cpp#L67](https://github.com/ros2/geometry2/blob/a1be44fdbd8c032351ffd3ddff8467d000f4c6d7/tf2_ros/src/tf2_echo.cpp#L67) fails. To fix this one would probably only need to use `rclcpp::iinit_and_remove_ros_arguments` instead of `rclcpp::init` as in `f2_monitor`.

https://github.com/ros2/geometry2/blob/a1be44fdbd8c032351ffd3ddff8467d000f4c6d7/tf2_ros/src/tf2_echo.cpp#L64
https://github.com/ros2/geometry2/blob/a1be44fdbd8c032351ffd3ddff8467d000f4c6d7/tf2_ros/src/tf2_monitor.cpp#L267

clalancette, 2021-04-29:
> To fix this one would probably only need to use `rclcpp::iinit_and_remove_ros_arguments` instead of `rclcpp::init` as in `f2_monitor`.

Yep, agreed.  We'll put this on the TODO list, but if you have time to submit a pull request, that's the fastest way to get this in.  Thanks.

PGotzmann, 2021-04-30:
> Yep, agreed. We'll put this on the TODO list, but if you have time to submit a pull request, that's the fastest way to get this in.

Thanks for the fast reply. I have created a PR with this small changes and also a backport to `foxy`. 
Let me know if there is anything that needs changing.

clalancette, 2021-04-30:
> Thanks for the fast reply. I have created a PR with this small changes and also a backport to `foxy`.

Thanks!

closed 2021-05-03 by clalancette

clalancette, 2021-05-03:
> @clalancette I would really love to see this backported to foxy, since that's the distribution I am currently using.
> How do you like me to proceed with #409?

My suggestion is that we just close #409, and I can use the Mergify bot to request backports to both Galactic and Foxy.  We can't merge the Galactic change yet (we are still in freeze for it), but we can at least have it open and ready for patch release 1.

PR #408 (2021-05-03): Fix tf2_echo does not work with ros-args (#407)
Enables the usage of `--ros-args`  by removing them before parsing the other command line parameters.
Fixes #407