ros2/geometry2 #421: Configure buffer cache time at runtime

jacobperron, 2021-05-23:
## Feature request

#### Feature description

We can configure the [cache time](https://github.com/ros2/geometry2/blob/cf612728c77f6c671e1534556f16b75cb352b53c/tf2/include/tf2/buffer_core.h#L99) when constructing the buffer object, however for applications that want to increase/decrease the cache time in third-party code, it would be nice to provide a canonical ROS interface for changing it.

#### Implementation considerations

Since we are already passing ROS node to the buffer, I think it would be fairly easy to declare a ROS parameter (e.g. `tf_buffer/cache_time`) that could be used to set the cache time at runtime:

https://github.com/ros2/geometry2/blob/cf612728c77f6c671e1534556f16b75cb352b53c/tf2_ros/include/tf2_ros/buffer.h#L75

If the user creates multiple tf2 buffers with the same ROS node, then I think having a single parameter set the cache for all buffers would be reasonable behavior. Otherwise, we need a way to distinguish different buffers.

We might also want to consider allowing updates *during* runtime. I.e. after the buffer has been created, let users change the cache time. I don't know if there would be an negative consequences for allowing the cache time to change after the buffer contains data. If we want to go this route, then we should create a setter method for this member:

https://github.com/ros2/geometry2/blob/cf612728c77f6c671e1534556f16b75cb352b53c/tf2/include/tf2/buffer_core.h#L331

clalancette, 2021-05-24:
Can you explain a little more of the rationale for changing the buffer cache time at runtime (vs at startup time)?  While we certainly can make it runtime configurable, I think there will be a lot of details that will make it hard to get correct.  So I'd like to see if there is something else we can do here.

jacobperron, 2021-05-24:
Perhaps _runtime_ in the title is a misnomer. I am proposing we make the cache time configurable at startup time (at the very least). I'm not actually sure if we want to make the cache time configurable during runtime, I just figured we should bring it up since if we expose a ROS parameter, we should make sure to handle someone trying to set it after startup.