ros2/geometry2 #433: Adding namespace to tf_listener and tf_broadcaster

MartiBolet, 2021-06-22:
Hi all,

We are now facing some trouble when working with multiple robots. `/tf` and `/tf_static` are now absolute path without namespace. This makes the multi-robot tasks difficult. Is there any possibility to solve this? We are now remapping this when needed but not sure if this is the proper way. [`--ros-args -r /tf:=tf -r /tf_static:=tf_static`]

When creating a transform broadcaster or listener it is also using absolute path by default. When we use 'rviz' with multiple robots within namespace, there's no way to visualize correctly without adding the same namespace of the robot to 'rviz' (`__ns:=...`).

It would be useful to have a parameter to instantiate the tf_broadcaster and tf_listener with a namespace (empty string by default) for the topic path.
For example, in `transform_listener.h`:
```c++
  template<class NodeT, class AllocatorT = std::allocator<void>>
  void init(
    NodeT && node,
    bool spin_thread,
    const rclcpp::QoS & qos,
    const rclcpp::QoS & static_qos,
    const std::string & ns = "",  // Added parameter, null by default
    const rclcpp::SubscriptionOptionsWithAllocator<AllocatorT> & options =
      rclcpp::SubscriptionOptionsWithAllocator<AllocatorT>())
  {
    using callback_t = std::function<void(tf2_msgs::msg::TFMessage::SharedPtr)>;
    callback_t cb = std::bind(&TransformListener::subscription_callback, this, std::placeholders::_1, false);
    callback_t static_cb = std::bind(&TransformListener::subscription_callback, this, std::placeholders::_1, true);

    // Added lines
    std::string tf_topic = "/tf";
    std::string tf_static_topic = "/tf_static";

    if (!ns.empty()) {
      std::cout << "Creating TF listener with ns: " << ns << std::endl;
      tf_topic = "/"+ns+"/tf";
      tf_static_topic = "/"+ns+"/tf_static";
    }
    // ----------

    message_subscription_tf_ = rclcpp::create_subscription<tf2_msgs::msg::TFMessage>(
      node,
      tf_topic,  // Replace the '/tf' to new var
      qos,
      std::move(cb),
      options);
    message_subscription_tf_static_ = rclcpp::create_subscription<tf2_msgs::msg::TFMessage>(
      node,
      tf_static_topic, // Replace the '/tf_static' to new var
      static_qos,
      std::move(static_cb),
      options);

    if (spin_thread) {
      initThread(node->get_node_base_interface());
    }
  }
```

Does it make sense? Any feedback is appreciate it.

Thanks in advance,
MartiBolet

closed 2021-06-23 by clalancette

arshadlab, 2022-08-09:
This is still an issue in Gazebo for tf's published from multiple plugins in multirobot scenario. e.g ros_diff_drive.  I think giving an explicit namespace to tf listener/broadcaster is good.

swiz23, 2022-12-23:
I've been wondering about best practice for this. Is it better to:

1. Keep the /tf and /tf_static topics in the root namespace, and just change the prefixes for all of the robots (in the URDF or via the robot_state_publisher `tf_prefix` param) so there are no tf collisions.
2. Remap the /tf and /tf_static topics into a robot specific namespace, and not even bother prefixing the tfs in the URDF or using the robot_state_publisher `tf_prefix` param.