ros2/geometry2 #435: tf2_py  TransformException catches all other exceptions

charlielito, 2021-07-07:
## Bug report

**Required Info:**

- Operating System:
  - Docker + Ubuntu 20.04
- Installation type:
  - From binaries
- Version or commit hash:
  - Foxy
- DDS implementation:
  - Fast-RTPS,
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue
It seems that `tf2_py.TransformException` can catch all others, since raising any exception and catching it with `tf2_py.TransformException` works. See example for better understanding.

```python
import tf2_py as tf2
try:
    raise tf2.TimeoutException("test")
except tf2.TransformException:
    print("catched")
```

#### Expected behavior
Different exceptions should raise.

#### Actual behavior
catching with `TransformException` works for other `tf2_ros` exceptions.

Is this the expected behavior?

clalancette, 2021-07-07:
Yes, this is the expected behavior.

The way this works is that there is a [TransformException](https://github.com/ros2/geometry2/blob/c36999b17cbf1e64a7937232133db79898bfa608/tf2/include/tf2/exceptions.h#L58) base class that is never directly thrown.  All of the actual exceptions are subclasses of that base class (e.g. [TimeoutException](https://github.com/ros2/geometry2/blob/c36999b17cbf1e64a7937232133db79898bfa608/tf2/include/tf2/exceptions.h#L133)).

What this allows is for the end-user to choose how to deal with exceptions.  If they only want to deal with the TimeoutException, they can catch only `TimeoutException`.  If they want to deal with all exceptions raised by TF2 (but not anything else in the Python ecosystem), then they can catch `TransformException`.  If they want to catch all exceptions, then they can catch `Exception`.

Given that this is the expected behavior, I'm going to close this issue out.

closed 2021-07-07 by clalancette

charlielito, 2021-07-07:
@clalancette  thanks for fast response. I just asked because when using the `BufferClient`, it never raises the exceptions, so the user never knows what the error is, for example when the server times out or there is any other error, exactly when using the `can_transform` method