ros2/geometry2 #438: rclpy BufferClient weird timeout on lookup_transform

charlielito, 2021-07-07:
## Bug report

**Required Info:**

- Operating System:
  - Docker + Ubuntu 20.04
- Installation type:
  - From binaries
- Version or commit hash:
  - Foxy
- DDS implementation:
  - Fast-RTPS,
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue
Calling `BufferClient.lookup_transform` always timeouts although if I call the buffer server via ros2 cli it works. 

Create a node like the following:

```python
#!/usr/bin/env python3
import rclpy
from rclpy.node import Node
from tf2_ros.buffer_client import BufferClient
from rclpy.duration import Duration
from std_msgs.msg import Empty

class TestNode(Node):
    def __init__(
        self,
    ):
        super().__init__("test_node")
        self.cb_group = None
        self.buffer = BufferClient(
            self,
            "/tf2_buffer_server",
            timeout_padding=1.0,
            callback_group=self.cb_group,
        )
        self.sub = self.create_subscription(
            msg_type=Empty,
            topic="/test_buffer_client",
            callback=self.buffer_cb,
            qos_profile=1,
        )

    def buffer_cb(self, msg):
        _ = self.buffer.lookup_transform(
            "map",
            "odom",
            rclpy.time.Time(seconds=0, nanoseconds=0),
            timeout=Duration(seconds=1.0),
        )
        self.get_logger().info("Success")

def main(args=None):
    rclpy.init(args=args)
    node = TestNode()
    executor = None
    rclpy.spin(node, executor)

if __name__ == "__main__":
    main()
```

Make sure the buffer server is already running and there is a valid transform between map->odom. Then publish to the topic that calls `BufferClient.lookup_transform`.

```
ros2 topic pub -1 /test_buffer_client std_msgs/msg/Empty
```

The traceback is:
```
Traceback (most recent call last):
  File "/workspace/rover/ros2/install/navigation/lib/navigation/navigation_test", line 33, in <module>
    sys.exit(load_entry_point('navigation', 'console_scripts', 'navigation_test')())
  File "/workspace/rover/ros2/build/navigation/navigation/node_test.py", line 53, in main
    rclpy.spin(node, executor)
  File "/opt/ros/foxy/lib/python3.8/site-packages/rclpy/__init__.py", line 191, in spin
    executor.spin_once()
  File "/opt/ros/foxy/lib/python3.8/site-packages/rclpy/executors.py", line 711, in spin_once
    raise handler.exception()
  File "/opt/ros/foxy/lib/python3.8/site-packages/rclpy/task.py", line 239, in __call__
    self._handler.send(None)
  File "/opt/ros/foxy/lib/python3.8/site-packages/rclpy/executors.py", line 426, in handler
    await call_coroutine(entity, arg)
  File "/opt/ros/foxy/lib/python3.8/site-packages/rclpy/executors.py", line 351, in _execute_subscription
    await await_or_execute(sub.callback, msg)
  File "/opt/ros/foxy/lib/python3.8/site-packages/rclpy/executors.py", line 118, in await_or_execute
    return callback(*args)
  File "/workspace/rover/ros2/build/navigation/navigation/node_test.py", line 33, in buffer_cb
    _ = self.buffer.lookup_transform(
  File "/opt/ros/foxy/lib/python3.8/site-packages/tf2_ros/buffer_client.py", line 87, in lookup_transform
    return self.__process_goal(goal)
  File "/opt/ros/foxy/lib/python3.8/site-packages/tf2_ros/buffer_client.py", line 201, in __process_goal
    raise tf2.TimeoutException("The LookupTransform goal sent to the BufferServer did not come back in the specified time. Something is likely wrong with the server")
tf2.TimeoutException: The LookupTransform goal sent to the BufferServer did not come back in the specified time. Something is likely wrong with the server
```

#### Expected behavior
It shouldn't crash since it works with the ros2 cli. If you use the following it works:
```
ros2 action send_goal /tf2_buffer_server tf2_msgs/action/LookupTransform '{target_frame: map, source_frame: odom}'
```

#### Actual behavior
It timeouts every time.

### Additional notes
I modified https://github.com/ros2/geometry2/blob/ros2/tf2_ros_py/tf2_ros/buffer_client.py#L61 to accept also a callback group that is passed to the `ActionClient`. Then, in the `TestNode` example above I set the `self.cb_group=ReentrantCallbackGroup()` and the executor to be a `MultiThreadedExecutor`. Only that way it works.

I am missing something?

charlielito, 2021-07-09:
I also having the same issue in `rclpp`, when using this inside a callback the `BufferClient.lookupTransform` timeouts

karl-schulz, 2021-08-12:
I have the same problem - I think this feature is actually not ready to be used?

clalancette, 2021-08-12:
> I am also having the same issue in `rclpp`, when using this inside a callback the `BufferClient.lookupTransform` timeouts.
> @clalancette I see there are some tests for this but not for testing it inside a callback function.

For better or worse, that is by design.  The problem with doing this is that you are blocking the executor, but the executor needs to run in order to get the transform.

The solution here is to make sure to never run blocking functions like `lookupTransform` from within a callback, or you use the MultiThreadedExecutor, or you instantiate the `TransformListener` with `spin_thread = True`.

karl-schulz, 2021-08-12:
Thanks! That makes sense, however is the usage of a separate thread in the `BufferClient` necessary in that case? We could use event.wait with a timeout instead of starting a new thread to check for a timeout, right?
https://docs.python.org/3/library/threading.html#threading.Event.wait

charlielito, 2021-08-20:
> For better or worse, that is by design. The problem with doing this is that you are blocking the executor, but the executor needs to run in order to get the transform.
>
> The solution here is to make [...]

`BufferClient` does have any `TransformListener`. And using `MultiThreadedExecutor` does not work, you have to pass a `ReentrantCallbackGroup` to the client inside the `BufferClient` class -which currently is not exposed-, as I already mentioned in the issue.

padhupradheep, 2021-09-20:
Is there any update on the issue? I'm also facing the same issue, but it is with rclcpp!

Flova, 2022-06-27:
This could be fixed by spinning a separate callback group in a separate thread. Independend execution of callback groups is currently only supported in rclcpp, but this pr aimed at implementing them https://github.com/ros2/rclpy/pull/909.
TF2 also features a not so clean fix. You can pass the TransformListener a newly created node object. This new node is used to gather the transforms in parallel if you also set spin_thread to True.

charlielito, 2022-06-28:
I think there is a misunderstanding here. The issue described in this thread refers to the [`BufferClient` class](https://github.com/ros2/geometry2/blob/ros2/tf2_ros_py/tf2_ros/buffer_client.py) which has nothing to do with a `TransformListener`. The `BufferClient` is just wrapping a `rclpy.action.client.ActionClient` to call a `tf2_ros buffer_server` using that client.

Flova, 2022-06-28:
Oh nvm. I missread this.

tigelbri, 2022-09-09:
I do have the same problem right now. Do I really need to change the BufferClient Interface or is there an other way of handling this issue?

bobbleballs, 2022-12-09:
Hi all, thanks very much for your input. I ran into this issue and I've found a solution that doesn't require changing the BufferClient interface, though I do think this should be changed.
A quick and dirty way is to change the Node's default callback group to a ReentrantCallbackGroup() then using a multithreaded excecutor to run the node. This way when you create the BufferClient it will use a ReentrantCallbackGroup by default.
Some example code below:
```
class ExampleActionServer(Node):
      def __init__(self):
          super().__init__('example_action_server')
          # Set the nodes default callback group to ReentrantCallbackGroup()
          self._default_callback_group = ReentrantCallbackGroup()
          self.tf_buffer = BufferClient(self, "/tf_action")
          self.tf_buffer.action_client.wait_for_server(5.0)
  
          self._action_server = ActionServer(
              self,
              ExampleAction,
              'example_action',
              self.execute_callback)

def main():
    rclpy.init()
    excecutor = rclpy.executors.MultiThreadedExecutor()
    
    example_action_server = ExampleActionServer()
    
    excecutor.add_node(example_action_server)
    excecutor.spin()
```
I've found that the buffer client does work from a callback this way.

tonynajjar, 2022-12-13:
This blog is quite relevant for this issue if you want to understand the issue; https://karelics.fi/deadlocks-in-rclpy/

I would propose to the maintainers to put the service client of the BufferClient in a separate callback group, like is done for the [subscribers of the transform listener](https://github.com/pixel-robotics/geometry2/blob/6530c38d5d52fb145940a04e65f7503c03391961/tf2_ros_py/tf2_ros/transform_listener.py#L82)

reinzor, 2024-02-22:
For people running into this problem:

Setting `spin_thread` to `true` did not fix my problem of doing `transform` lookups within a callback. Adding a `rclpy.executors.MultiThreadedExecutor` did.

Flova, 2024-03-18:
Also, for people that have timeout/blocking issues in ROS2 in general, the multithreaded executor helps. But keep in mind that the default callback group is mutually exclusive and you need to select a reentrant callback group (or multiple mutually exclusive ones) for the involved callbacks to use multiple threads in the multithreaded executor.

zacharyyamaoka, 2024-12-21:
I was getting poor performance using a tf_buffer with multithreaded executors, as the high frequency TF data seemed to overwhelm all the other callback groups.

Here is a log where I do a self.node.create_rate(100).sleep() in an action server while I wait for a condition to be true. I would expect the loop time to be consistent 0.01s, but it has trouble keeping up.
```
[INFO] [1734795460.901186591] [bam_BAMGPU.deltax_server]: Recivied Action Request
[INFO] [1734795460.902852478] [bam_BAMGPU.deltax_server]: Executing goal...
[INFO] [1734795460.904370023] [bam_BAMGPU.deltax_server]: Waiting. . .
[INFO] [1734795460.905848492] [bam_BAMGPU.deltax_server]: Loop time: 0.001
0.2872281323269015 0.0
[INFO] [1734795460.908104775] [bam_BAMGPU.deltax_server]: Waiting. . .
[INFO] [1734795460.909355380] [bam_BAMGPU.deltax_server]: Loop time: 0.004
0.2872281323269015 0.0
[INFO] [1734795460.913589135] [bam_BAMGPU.deltax_server]: Waiting. . .
[INFO] [1734795460.920546263] [bam_BAMGPU.deltax_server]: Loop time: 0.005
0.2872281323269015 0.0
[INFO] [1734795460.924742035] [bam_BAMGPU.deltax_server]: Waiting. . .
[INFO] [1734795460.929696233] [bam_BAMGPU.deltax_server]: Loop time: 0.011
0.2872281323269015 0.0
[INFO] [1734795460.934394203] [bam_BAMGPU.deltax_server]: Waiting. . .
[INFO] [1734795460.940699708] [bam_BAMGPU.deltax_server]: Loop time: 0.01
0.2872281323269015 0.0
[INFO] [1734795460.944343166] [bam_BAMGPU.deltax_server]: Waiting. . .
[INFO] [1734795460.944838221] [bam_BAMGPU.deltax_server]: Loop time: 0.01
0.2870122760440745 0.0
[INFO] [1734795461.850687009] [bam_BAMGPU.deltax_server]: Waiting. . .
[INFO] [1734795462.905678193] [bam_BAMGPU.deltax_server]: Loop time: 0.906
7.141428428546883e-05 0.0
```

Removing TF from the script fixes the issue, and the loop time is a consistent 0.01s. Here is the problematic code:
  
```
self.tf_buffer = Buffer()
self.tf_listener = TransformListener(self.tf_buffer, self, spin_thread=False)
```

After searching online I found this article: https://robotics.stackexchange.com/questions/99416/transformlistener-x10-higher-cpu-load-in-rclpy-than-rclpp , which lead me to here! Instead of using a TransformListenter, I am now using the BufferClient

```
# self.tf_buffer = Buffer()
# self.tf_listener = TransformListener(self.tf_buffer, self, spin_thread=False) 
self.tf_buffer_client = BufferClient(self, f"{self.get_namespace()}/tf2_buffer_server", check_frequency=100)
```

And launching the BufferServer

```
    tf_buffer_server = Node(
    package="tf2_ros",
    executable="buffer_server",
    output="both",
    emulate_tty=True
    )
```

As others described, I can query the server from ros cli

```
ros2 action send_goal /tf2_buffer_server tf2_msgs/action/LookupTransform '{target_frame: base_link, source_frame: eef_link}'
```

I was able to connect to the server
```self.tf_buffer_client.action_client.wait_for_server(5)```

However, I could not get a transform! It would just time out

```
self.tf_buffer_client.lookup_transform(self.planning_frame, self.eef_frame,rclpy.time.Time())
```

I have edited the __process_goal() function in buffer_client.py to use a rate.sleep() instead. I also just call the lookup_transform  within another action callback. If I call lookup_transform during the node __init__() it will block as it's sleeping on the main thread.

```
def __process_goal(self, goal: LookupTransformGoal) -> TransformStamped:
# TODO(sloretz) why is this an action client? Service seems more appropriate.
if not self.action_client.server_is_ready():
    raise tf2.TimeoutException("The BufferServer is not ready.")

send_goal_future = self.action_client.send_goal_async(goal)

clock = Clock()
start_time = clock.now()
timeout = Duration.from_msg(goal.timeout)
timeout_padding = self.timeout_padding

while not send_goal_future.done():
    if clock.now() > start_time + timeout + timeout_padding:
        break
    self.wait_rate.sleep()

# This shouldn't happen, but could in rare cases where the server hangs
if not send_goal_future.done():
    raise tf2.TimeoutException("The LookupTransform goal sent to the BufferServer did not come back in the specified time. Something is likely wrong with the server.")

# Raises if future was given an exception
goal_handle = send_goal_future.result()

if not goal_handle.accepted:
    raise tf2.TimeoutException("The LookupTransform goal sent to the BufferServer did not come back with accepted status. Something is likely wrong with the server.")

response = self.action_client._get_result(goal_handle)

return self.__process_result(response.result)
```

How I use it

```
class ActionServer(Node):
   
    def __init__(self):
        super().__init__('my_action_server')

        self.action_cb_group = ReentrantCallbackGroup()
        self.action_server = ActionServer(self, RobotAPI, 'robot_api/request', self.action_callback, callback_group=self.action_cb_group)
        
        self.result = RobotAPI.Result()
        self.waiting_rate = self.create_rate(100)

        self.tf_buffer_client = BufferClient(self, f"{self.get_namespace()}/tf2_buffer_server", check_frequency=100)
        
        self.planning_frame = 'base_link'
        self.eef_frame = 'xyz_link'
        self.eef_target_frame = 'eef_target'

        if not self.tf_buffer_client.action_client.wait_for_server(5):
            self.get_logger().error(f"Could not connect to TF buffer server")

    def action_callback(self, goal_handle):

        goal_handle.succeed()

        self.request = goal_handle.request
        
        while True:
            self.waiting_rate.sleep()

            t_eef = self.tf_buffer_client.lookup_transform(self.planning_frame, self.eef_frame,rclpy.time.Time())
            t_eef = t_eef.transform

            pos_diff, rot_diff = pose_difference(t_eef.translation, t_eef.rotation, t_eef_target.translation, t_eef_target.rotation)

            if pos_diff < 0.005:
                break

        return self.result 
       
```

Flova, 2024-12-22:
We now use this hack and it works also really well:
https://github.com/bit-bots/bitbots_main/tree/main/bitbots_misc/bitbots_tf_buffer

It essentially wraps a rclcpp Node, TF Buffer and Listener in a wrapper that matches the Python ft2.BufferInterface. It is significantly more performant, as all communication is done in cpp and the transform calls are performed directly on the cpp node via shred memory. The only downside is the additional node. 

Additionally, I started porting the events executor to native Python rclpy and got a 10x increased throughput with a quarter of the CPU utilization. But this will take a while to reach production readiness.

mbpeterson70, 2025-04-05:
This tip from @reinzor worked for me:

> Setting `spin_thread` to `true` did not fix my problem of doing `transform` lookups within a callback. Adding a `rclpy.executors.MultiThreadedExecutor` did.

To add a small detail, for me it was just switching this line in `main` from:

```python
rclpy.spin(node)
```

to

```python
from rclpy.executors import MultiThreadedExecutor

...

executor = MultiThreadedExecutor()
executor.add_node(node)
executor.spin()
```

Flova, 2025-04-06:
You also need to provide an appropriate amount of threads, otherwise it still deadlocks.

Flova, 2025-04-06:
Imo a proper async wait for transforms with a timeout that can be awaited would be the way to go. The calling parent function would need to be async too, but rclpy does support async callbacks. This would allow the executor to spin freely during the wait period allowing it to process incoming data eliminateing the deadlock. 

As stated earlier https://github.com/bit-bots/bitbots_tf_buffer is a drop in replacement that also solves this problem, while being also significantly faster.

charlielito, 2025-04-08:
> Imo a proper async wait for transforms with a timeout that can be awaited would be the way to go. The calling parent function would need to be async too, but rclpy does support async callbacks. This [...]

I agree. A proper AsyncBufferClient should be available to the users. I ported the implementation to use th async api as follows:

```python
#! /usr/bin/python

import tf2_py as tf2
import tf2_ros
import rclpy
import rclpy.task

from rclpy.action.client import ActionClient
from rclpy.duration import Duration
from rclpy.node import Node
from tf2_msgs.action import LookupTransform

async def future_with_timeout(
    node: Node, future: rclpy.task.Future, timeout_sec: float
):
    """!
    function to await a future with a given timeout
    :param node: The ROS2 node.
    :param future: The future to await.
    :param timeout_sec: The timeout.
    :return: The future's result.
    """
    first_done_future = rclpy.task.Future()

    def handler(arg=None):
        first_done_future.set_result(None)

    timer = node.create_timer(timeout_sec, handler)
    future.add_done_callback(handler)

    await first_done_future

    if not future.done():
        raise TimeoutError(f"Future timed out after {timeout_sec} seconds")

    timer.cancel()
    timer.destroy()

    return future.result()

class AsyncBufferClient(tf2_ros.BufferInterface):
    """
    Async action client-based implementation of BufferInterface.
    """

    def __init__(
        self,
        node,
        ns,
        timeout_padding=Duration(seconds=2.0),
        callback_group=None,
    ):
        """
        .. function:: __init__(ns, timeout_padding = rospy.Duration.from_sec(2.0))

            Constructor.

            :param node: The ROS2 node.
            :param ns: The namespace in which to look for a BufferServer.
            :param timeout_padding: A constant timeout to add to blocking calls.
            :param callback_group: The callback group to use for the action client.
        """
        tf2_ros.BufferInterface.__init__(self)
        self.node = node
        self.action_client = ActionClient(
[... 205 lines cut ...]
        self.sub = self.create_subscription(
            msg_type=Empty,
            topic="/test_buffer_client",
            callback=self.buffer_cb,
            qos_profile=1,
            callback_group=ReentrantCallbackGroup(),
        )

    async def buffer_cb(self, msg):
        result = await self.buffer.lookup_transform(
            "map",
            "odom",
            rclpy.time.Time(seconds=0, nanoseconds=0),
            timeout=Duration(seconds=0.1),
        )
        self.get_logger().info(f"Success: {result}")

def main(args=None):
    rclpy.init(args=args)
    node = TestNode()
    rclpy.spin(node)

if __name__ == "__main__":
    main()
```

charlielito, 2025-04-08:
> We now use this hack and it works also really well: https://github.com/bit-bots/bitbots_tf_buffer
> 
> It essentially wraps a rclcpp Node, TF Buffer and Listener in a wrapper that matches the Python ft2.BufferInterface. It is significantly more performant, as all communication is done in cpp and the transform calls are performed directly on the cpp node via shred memory. The only downside is the additional node.
> 
> Additionally, I started porting the events executor to native Python rclpy and got a 5x increased throughput with a quarter of the CPU utilization. But this will take a while to reach production readiness.

@Flova I am curious if you finished the port of the events executor to native python and where can we find it?

Flova, 2025-04-08:
> I agree. A proper AsyncBufferClient should be available to the users. I ported the implementation to use the async api as follows:

Look very cool. I would be glad if we could get something like this upstream with a PR. I also think that the "normal" non service client buffer could very much benefit from this.  

> [@Flova](https://github.com/Flova) I am curious if you finished the port of the events executor to native python and where can we find it?

I wan't the only one working on an implementation of the events executor for Python and somebody finished a more polished version first. It has been added to rolling in https://github.com/ros2/rclpy/pull/1391 and I have backported this implementation to jazzy in https://github.com/ros2/rclpy/pull/1435.