ros2/geometry2 #440: Reduce TransformListener nodes

gezp, 2021-07-29:
hi, I'm working for [Summer Student Program of Navigation2 ](https://navigation.ros.org/2021summerOfCode/projects/spinners.html) :  reduce the number of nodes running in Nav2 ([tickets in Nav2](https://github.com/ros-planning/navigation2/issues/816)), due to Node is overhead in ROS2.

When i [reduce amcl_node nodes](https://github.com/ros-planning/navigation2/pull/2483), i notice that the constructor of `TransformListener` has a argument `spin_thread` with default value true, which create a internal node to subscribe tf topic and a dedicated thread to spin this node.

i think the best way is to use new callback group and executor with dedicated thread so that we could avoid creating internal node which is overhead. 

* If spin_thread, then create new callback group, and use executor to spin callback group instead of node.
* If not spin_thread, then its on the application to spin (no change)

After implementing this feature, we can use `TransformListener` in Node (for example [AmclNode](https://github.com/ros-planning/navigation2/blob/main/nav2_amcl/include/nav2_amcl/amcl_node.hpp)) like this to avoid creating internal Node.

```c++
tf_buffer_ = std::make_shared<tf2_ros::Buffer>(get_clock());
tf_listener_ = std::make_shared<tf2_ros::TransformListener>(
   *tf_buffer_, shared_from_this(), true);
```

any discussion is welcome, and i will make a PR if this feature is good.

SteveMacenski, 2021-07-30:
Our goal is to make each server in Nav2 only have a single ROS node and making best practices use of executors. TF creates a node in the Listener which we argue should not be there and instead use an input node + an executor as needed.

clalancette, 2021-10-01:
Now that we've merged #442, can we close this one?

SteveMacenski, 2021-10-01:
Indeed

clalancette, 2021-10-01:
Thanks, closing.

closed 2021-10-01 by clalancette