ros2/geometry2 #443: canTransform target_frame - frame does not exist

boyali, 2021-08-02:
## Bug report

**Required Info:**

- Operating System:
  - <Ubuntu 20.04>
- Installation type:
  - 
- DDS implementation:
  - 
- Client library (if applicable):
  - < rclcpp ->

#### Steps to reproduce issue

```
Followed this tutorial, 
https://docs.ros.org/en/galactic/Tutorials/tf2.html

ros2 run tf2_ros static_transform_publisher 1 2 3 0.5 0.1 -1.0 foo bar
ros2 run tf2_ros tf2_echo foo bar 

```

#### Expected behavior
At time 0.0
- Translation: [1.000, 2.000, 3.000]
- Rotation: in Quaternion [-0.475, -0.076, 0.240, 0.843]

#### Actual behavior
In ros2-foxy environment 
[INFO 1627906637.867571464] [tf2_echo]: Waiting for transform foo ->  bar: Invalid frame ID "foo" passed to canTransform argument target_frame - frame does not exist
[INFO 1627906639.867518947] [tf2_echo]: Waiting for transform foo ->  bar: Invalid frame ID "foo" passed to canTransform argument target_frame - frame does not exist
[INFO 1627906640.867562758] [tf2_echo]: Waiting for transform foo ->  bar: Invalid frame ID "foo" passed to canTransform argument target_frame - frame does not exist

#### Additional information

This error occurs in both of ros2-foxy and ros2-galactic.

clalancette, 2021-08-02:
This seems to work fine for me in Galactic.  Can you give some more information on exactly what versions of the ros2 packages you are using, as well as exact directions of what you did from the time of starting your terminal?

boyali, 2021-08-02:
Thank you very much for the prompt response. Here are the steps I followed. 
Ubuntu Version:
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.2 LTS
Release:	20.04
Codename:	focal

I installed ros2-galactic and ros2-foxy using Ubuntu packages, I also tried in the built version of ros-galactic and received the same error. 

Installation procedures:
ros2-foxy     :  sudo apt install ros-foxy-desktop
ros2-galactic: 
 - sudo apt install ros-galactic-desktop
 - built by following https://docs.ros.org/en/galactic/Installation/Ubuntu-Development-Setup.html

 Just after installations, 
 Followed this tutorial, 
https://docs.ros.org/en/galactic/Tutorials/tf2.html

ros2 run tf2_ros static_transform_publisher 1 2 3 0.5 0.1 -1.0 foo bar
ros2 run tf2_ros tf2_echo foo bar 

I followed these steps exactly. The results I am getting on my personal computer.

[INFO 1627912352.563685312] [tf2_echo]: Waiting for transform foo ->  bar: Invalid frame ID "foo" passed to canTransform argument target_frame - frame does not exist
[INFO 1627912354.563029606] [tf2_echo]: Waiting for transform foo ->  bar: Invalid frame ID "foo" passed to canTransform argument target_frame - frame does not exist
[INFO 1627912355.563316302] [tf2_echo]: Waiting for transform foo ->  bar: Invalid frame ID "foo" passed to canTransform argument target_frame - frame does not exist
[INFO 1627912357.563317360] [tf2_echo]: Waiting for transform foo ->  bar: Invalid frame ID "foo" passed to canTransform argument target_frame - frame does not exist
[INFO 1627912359.563094231] [tf2_echo]: Waiting for transform foo ->  bar: Invalid frame ID "foo" passed to canTransform argument target_frame - frame does not exist
[INFO 1627912361.563104046] [tf2_echo]: Waiting for transform foo ->  bar: Invalid frame ID "foo" passed to canTransform argument target_frame - frame does not exist
[INFO 1627912363.562983205] [tf2_echo]: Waiting for transform foo ->  bar: Invalid frame ID "foo" passed to canTransform argument target_frame - frame does not exist
[INFO 1627912364.563160038] [tf2_echo]: Waiting for transform foo ->  bar: Invalid frame ID "foo" passed to canTransform argument target_frame - frame does not exist
[INFO 1627912366.563112539] [tf2_echo]: Waiting for transform foo ->  bar: Invalid frame ID "foo" passed to canTransform argument target_frame - frame does not exist
[INFO 1627912368.563059722] [tf2_echo]: Waiting for transform foo ->  bar: Invalid frame ID "foo" passed to canTransform argument target_frame - frame does not exist
[INFO 1627912370.563019881] [tf2_echo]: Waiting for transform foo ->  bar: Invalid frame ID "foo" passed to canTransform argument target_frame - frame does not exist
[INFO 1627912371.563021169] [tf2_echo]: Waiting for transform foo ->  bar: Invalid frame ID "foo" passed to canTransform argument target_frame - frame does not exist
[INFO 1627912372.563057594] [tf2_echo]: Waiting for transform foo ->  bar: Invalid frame ID "foo" passed to canTransform argument target_frame - frame does not exist
[INFO 1627912373.563182707] [tf2_echo]: Waiting for transform foo ->  bar: Invalid frame ID "foo" passed to canTransform argument target_frame - frame does not exist

boyali, 2021-08-02:
Solved:
I think this is related to the following setting which solved the problem. 

If local_host_only setting is set to 0 it produces this error. Setting it to one removed the error. 
export ROS_LOCALHOST_ONLY=1

Thank you very much.

padhupradheep, 2021-09-17:
Sorry for hijacking the issue, but felt that the [question](https://answers.ros.org/question/386764/ros2-buffercantransform-always-returns-false/) I raised in ROS-Answers, is similar to this issue posted here!

closed 2021-09-24 by aprotyas