ros2/geometry2 #450: On windows, NO_ERROR in tf2/exceptions.h conflicts with NO_ERROR in winerror.h

dzywater, 2021-08-29:


clalancette, 2021-08-30:
This one is tricky.  I have a pull request open that works towards solving it here: https://github.com/ros2/geometry2/pull/349 .  There are two caveats with it:

1.  There still some bug on Windows that makes it not so happy.
2. Even with that PR applied, there is still going to be a conflict on Windows.

Number 2 is unavoidable; since we really want a deprecation cycle for it, we just have to bite the bullet and leave it in place until after Humble.  I just haven't gotten back to problem 1 to see if I can fix it.

aprotyas, 2021-09-18:
#456 also introduces a workaround for Foxy (which can be cherry-picked to Galactic).

aprotyas, 2021-09-24:
This can probably be closed now since #349, #456, #458 have all been merged.

clalancette, 2021-09-24:
Good call, I'll close this out.

closed 2021-09-24 by clalancette