ros2/geometry2 #477: Create a callback group to spin in Python TransformListener

jacobperron, 2021-11-16:
## Feature request

#### Feature description

Same as #442, but applied to the [Python implementation](https://github.com/ros2/geometry2/blob/16562cee8694c11f1f82b2bdbde2814fca1c7954/tf2_ros_py/tf2_ros/transform_listener.py#L44).

This will make the behavior consistent between languages.

#### Implementation considerations

It would be nice to emit a warning if the user is relying on the old behavior, but I'm not sure how feasible this is.
At the very least, we should add a release note to the Humble release page to highlight the behavior change.

gezp, 2021-11-16:
is anyone working for this? if not, i would like to continue this work for python implementation.

jacobperron, 2021-11-16:
I just took a brief look, but it doesn't look like the `rclpy` executor API supports callback groups (please correct me if I'm mistaken). I've opened a ticket upstream https://github.com/ros2/rclpy/issues/850.

gezp, 2021-11-16:
> but it doesn't look like the rclpy executor API supports callback groups

sorry, i didn't notice that rclpy API is some different of rlcpp before. you are right, we need some features from `rlcpy` .