ros2/geometry2 #482: tf2_tools view_frames.py produces no pdf and prints result to command line instead

hortenbach, 2021-12-07:
## Bug report

**Required Info:**

- Operating System:
Ubuntu 20.04.3
- Installation type:
source
- Version or commit hash:
foxy
- DDS implementation:

- Client library (if applicable):
tf2_tools

#### Steps to reproduce issue
I am following the Introduction to tf2 tutorial https://docs.ros.org/en/foxy/Tutorials/Tf2/Introduction-To-Tf2.html
And when I run the command from the "1 Using view_frames" section
```
ros2 run tf2_tools view_frames.py
```
I don't get a frames.pdf but text on command line instead. 
#### Expected behavior
view_frames.py should output a frames.pdf
#### Actual behavior
Result gets postes to command line
```
$ ros2 run tf2_tools view_frames.py
[INFO] [1638875729.962967886] [view_frames]: Listening to tf data during 5 seconds...
[INFO] [1638875734.963944893] [view_frames]: Generating graph in frames.pdf file...
[INFO] [1638875734.966703694] [view_frames]: Result:tf2_msgs.srv.FrameGraph_Response(frame_yaml="turtle2: \n  parent: 'world'\n  broadcaster: 'default_authority'\n  rate: 62.699\n  most_recent_transform: 1638875734.962492\n  oldest_transform: 1638875729.954437\n  buffer_length: 5.008\nturtle1: \n  parent: 'world'\n  broadcaster: 'default_authority'\n  rate: 62.699\n  most_recent_transform: 1638875734.962528\n  oldest_transform: 1638875729.954505\n  buffer_length: 5.008\n")

```

aprotyas, 2021-12-07:
In Foxy, `tf2_tools` does not have a runtime dependency on `graphviz`. So, if for some reason `graphviz` did not get installed in the system - because, say, this tutorial was the very first thing someone did - then this `dot` invocation (tool comes with `graphviz`) dies silently: https://github.com/ros2/geometry2/blob/babb5f982a17708e56bbd8236f4dc5ef5f98b47d/tf2_tools/scripts/view_frames.py#L78

As such, https://github.com/ros2/geometry2/pull/313/ does not have to be backported - but it should be, because of security reasons. Rather, the fix of listing `graphviz` as a runtime dependency was introduced in https://github.com/ros2/geometry2/pull/351/. Now, if this is being backported, `tf2_tools` might as well be synced to `ros2#HEAD` across all distros for now. Would that be possible?

aprotyas, 2021-12-08:
Thank you for the report @hortenbach! I'm closing this issue since #483 has been merged.

PR #483 (2021-12-08): [tf2_tools] Add `graphviz` as a runtime dependency
This PR is an alternative to backporting #351 since that requires
`tf2_ros_py`, among other difficulties.

Without this runtime dependency, the subprocess used to invoke the `dot`
utility that ships with `graphviz` silently dies, and a `frames.pdf`
file is not generated.

Closes #482.
closed 2021-12-08 by aprotyas