ros2/geometry2 #487: static_transform_publisher with transient_local doesn't work

KenYN, 2021-12-16:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - Galactic
- Version or commit hash:
  - Latest from apt-get
- DDS implementation:
  - CycloneDDS
- Client library (if applicable):
  - tf2_ros

#### Steps to reproduce issue
First, this is similar to #183, but the suggested resolution doesn't seem to work, as seen here; try the following two commands:

    $ ros2 run tf2_ros static_transform_publisher --x 1 --y 1 --z 1 --frame-id f1 --child-frame-id f2 &
    $ ros2 topic echo --qos-durability transient_local /tf_static

If I understand #183 correctly, the `--qos-durability transient_local` parameter will match up with the `static_transform_publisher`'s QoS.

#### Expected behavior
I expect to see the output:

```
transforms:
- header:
    stamp:
      sec: 1639634982
      nanosec: 150475886
    frame_id: map
  child_frame_id: base_link
  transform:
    translation:
      x: 1.0
      y: 1.0
      z: 1.0
    rotation:
      x: 0.0
      y: 0.0
      z: 0.0
      w: 1.0
```

#### Actual behavior
All I see is the below and the program hangs waiting for messages:

`1639638202.723132 [0]       ros2: selected interface "lo" is not multicast-capable: disabling multicast`

#### Additional information
Note that I am using `loopback` due to multicast being blocked by my router, but I don't think that is connected.

clalancette, 2021-12-16:
You also need to add `--qos-reliability reliable` in Galactic.  On Rolling, that is no longer needed.  Does that work for you?

KenYN, 2021-12-17:
Thanks, that fixed it!

closed 2021-12-17 by KenYN