ros2/geometry2 #503: :man_farmer: `tf2_ros_py` `test_buffer_client.py` segfaults on `release` and `fastrtps` nightlies

Blast545, 2022-02-03:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - Buildfarm, source
- Version or commit hash:
  - https://gist.github.com/Blast545/eb4b7b16ddf5e069df52de1b900f15dc
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue
Run a nightly linux release or fastrtps job. `Patient zero` cases can be seen here:
https://build.ros2.org/view/Rci/job/Rci__nightly-release_ubuntu_focal_amd64/587/
https://build.ros2.org/view/Rci/job/Rci__nightly-fastrtps_ubuntu_focal_amd64/589/

#### Expected behavior
All tests passes

#### Actual behavior
Some fail, in particular, `test_buffer_client.py` segfaults:
https://github.com/ros2/geometry2/blob/ros2/tf2_ros_py/test/test_buffer_client.py#L98

#### Additional information
There are no important differences between the repos file from the previous passing nightly in both of the cases.

When the segfaults the stacktrace shows these two lines:
https://github.com/ros2/geometry2/blob/210791c2a57a5c7c87e348e684c9cdc28469cb70/tf2_ros_py/test/test_buffer_client.py#L126
https://github.com/ros2/geometry2/blob/210791c2a57a5c7c87e348e684c9cdc28469cb70/tf2_ros_py/test/test_buffer_client.py#L130
So I would assume the problem is related to objects being destructed in a different order than the expected.

Blast545, 2022-02-03:
Trying to reproduce the error on `ci.ros2.org`, ci_linux release up to `tf2_ros_py`

[image]

Blast545, 2022-02-04:
Another instance: https://build.ros2.org/view/Rci/job/Rci__nightly-debug_ubuntu_focal_amd64/588/

Blast545, 2022-02-22:
It happened here as well: https://ci.ros2.org/view/nightly/job/nightly_linux_release/2199/

clalancette, 2022-02-22:
I've been attempting to track this down a bit.  I'm able to reproduce locally by building up to `tf2_ros_py` and then running the tests:

```
colcon build --event-handlers console_direct+ --packages-up-to tf2_ros_py
. install/setup.bash
py.test -s src/ros2/geometry2/tf2_ros_py/test
```

(it doesn't happen every time, but maybe one out of every 6 or 7 runs for me).  I then rebuilt `rmw_fastrtps_shared_cpp` with debug symbols:

```
colcon build --event-handlers console_direct+ --packages-select rmw_fastrtps_shared_cpp --cmake-args -DCMAKE_BUILD_TYPE=Debug
```

With that I was able to collect a coredump, and the stack trace looks like this:

```
(gdb) bt
#0  __GI_raise (sig=<optimized out>) at ../sysdeps/unix/sysv/linux/raise.c:50
#1  <signal handler called>
#2  0x0000000000000241 in ?? ()
#3  0x00007ff49369403b in rmw_fastrtps_shared_cpp::__rmw_wait (identifier=0x7ff49375cee3 "rmw_fastrtps_cpp", subscriptions=0x33db0a8, 
    guard_conditions=0x33db0c0, services=0x33db0f0, clients=0x33db0d8, events=0x33db108, wait_set=0x7ff4840014c0, wait_timeout=0x7ff48e03cca0)
    at /home/ubuntu/ros2_ws/src/ros2/rmw_fastrtps/rmw_fastrtps_shared_cpp/src/rmw_wait.cpp:198
#4  0x00007ff493754023 in rmw_wait () from /home/ubuntu/ros2_ws/install/rmw_fastrtps_cpp/lib/librmw_fastrtps_cpp.so
#5  0x00007ff4973ff232 in rmw_wait () from /home/ubuntu/ros2_ws/install/rmw_implementation/lib/librmw_implementation.so
#6  0x00007ff49640bb13 in rcl_wait () from /home/ubuntu/ros2_ws/install/rcl/lib/librcl.so
#7  0x00007ff496569537 in rclpy::WaitSet::wait(long) ()
   from /home/ubuntu/ros2_ws/install/rclpy/lib/python3.8/site-packages/rclpy/_rclpy_pybind11.cpython-38-x86_64-linux-gnu.so
#8  0x00007ff49656c94e in pybind11::cpp_function::cpp_function<void, rclpy::WaitSet, long, pybind11::name, pybind11::is_method, pybind11::sibling, char [48]>(void (rclpy::WaitSet::*)(long), pybind11::name const&, pybind11::is_method const&, pybind11::sibling const&, char const (&) [48])::{lambda(rclpy::WaitSet*, long)#1}::operator()(rclpy::WaitSet*, long) const ()
   from /home/ubuntu/ros2_ws/install/rclpy/lib/python3.8/site-packages/rclpy/_rclpy_pybind11.cpython-38-x86_64-linux-gnu.so
#9  0x00007ff4965723db in void pybind11::detail::argument_loader<rclpy::WaitSet*, long>::call_impl<void, pybind11::cpp_function::cpp_function<void, rclpy::WaitSet, long, pybind11::name, pybind11::is_method, pybind11::sibling, char [48]>(void (rclpy::WaitSet::*)(long), pybind11::name const&, pybind11::is_method const&, pybind11::sibling const&, char const (&) [48])::{lambda(rclpy::WaitSet*, long)#1}&, 0ul, 1ul, pybind11::detail::void_type>(pybind11::cpp_function::cpp_function<void, rclpy::WaitSet, long, pybind11::name, pybind11::is_method, pybind11::sibling, char [48]>(void (rclpy::WaitSet::*)(long), pybind11::name const&, pybind11::is_method const&, pybind11::sibling const&, char const (&) [48])::{lambda(rclpy::WaitSet*, long)#1}&, std::integer_sequence<unsigned long, 0ul, 1ul>, pybind11::detail::void_type&&) () from /home/ubuntu/ros2_ws/install/rclpy/lib/python3.8/site-packages/rclpy/_rclpy_pybind11.cpython-38-x86_64-linux-gnu.so
#10 0x00007ff49656ff64 in std::enable_if<std::is_void<void>::value, pybind11::detail::void_type>::type pybind11::detail::argument_loader<rclpy::WaitSet*, long>::call<void, pybind11::detail::void_type, pybind11::cpp_function::cpp_function<void, rclpy::WaitSet, long, pybind11::name, pybind11::is_method, pybind11::sibling, char [48]>(void (rclpy::WaitSet::*)(long), pybind11::name const&, pybind11::is_method const&, pybind11::sibling const&, char const (&) [48])::{lambda(rclpy::WaitSet*, long)#1}&>(pybind11::cpp_function::cpp_function<void, rclpy::WaitSet, long, pybind11::name, pybind11::is_method, pybind11::sibling, char [48]>(void (rclpy::WaitSet::*)(long), pybind11::name const&, pybind11::is_method const&, pybind11::sibling const&, char const (&) [48])::{lambda(rclpy::WaitSet*, long)#1}&) && () from /home/ubuntu/ros2_ws/install/rclpy/lib/python3.8/site-packages/rclpy/_rclpy_pybind11.cpython-38-x86_64-linux-gnu.so
#11 0x00007ff49656ef87 in pybind11::cpp_function::initialize<pybind11::cpp_function::initialize<void, rclpy::WaitSet, long, pybind11::name, pybind11::is_method, pybind11::sibling, char [48]>(void (rclpy::WaitSet::*)(long), pybind11::name const&, pybind11::is_method const&, pybind11::sibling const&, char const (&) [48])::{lambda(rclpy::WaitSet*, long)#1}, void, rclpy::WaitSet*, long, pybind11::name, pybind11::is_method, pybind11::sibling, char [48]>(pybind11::cpp_function::initialize<void, rclpy::WaitSet, long, pybind11::name, pybind11::is_method, pybind11::sibling, char [48]>(void (rclpy::WaitSet::*)(long), pybind11::name const&, pybind11::is_method const&, pybind11::sibling const&, char const (&) [48])::{lambda(rclpy::WaitSet*, long)#1}&&, void (*)(rclpy::WaitSet*, long), pybind11::name const&, pybind11::is_method const&, pybind11::sibling const&, char const (&) [48])::{lambda(pybind11::detail::function_call&)#3}::operator()(pybind11::detail::function_call) const () from /home/ubuntu/ros2_ws/install/rclpy/lib/python3.8/site-packages/rclpy/_rclpy_pybind11.cpython-38-x86_64-linux-gnu.so
#12 0x00007ff49656f04d in pybind11::cpp_function::initialize<pybind11::cpp_function::initialize<void, rclpy::WaitSet, long, pybind11::name, pybind11::is_method, pybind11::sibling, char [48]>(void (rclpy::WaitSet::*)(long), pybind11::name const&, pybind11::is_method const&, pybind11::sibling const&, char const (&) [48])::{lambda(rclpy::WaitSet*, long)#1}, v [... cut ...] hon3.8/site-packages/rclpy/_rclpy_pybind11.cpython-38-x86_64-linux-gnu.so
#13 0x00007ff49644649d in pybind11::cpp_function::dispatcher(_object*, _object*, _object*) ()
```

(I removed the last lines for brevity).  It seems like some kind of issue with `rmw_fastrtps` both because of the stack trace above, and because running with `rmw_cyclonedds_cpp` doesn't ever crash.

I'm going to continue to poke at this, but @MiguelCompany @JaimeMartin any help you can provide here would be most appreciated.

clalancette, 2022-02-23:
All right, I did some additional debugging here and enabled ASan.  This looks like it may be a bug in `rclpy`, though I'm not 100% certain yet.

Looking at https://github.com/ros2/rclpy/blob/master/rclpy/src/rclpy/wait_set.cpp , it is clear that we create an `rcl_wait_set`, populate it, and then call `rcl_wait` on it with no locking.  Further, in https://github.com/ros2/rclpy/blob/master/rclpy/src/rclpy/wait_set.cpp , we explicitly drop the GIL, so we don't have that protection either.  With all of that, it looks like it can be the case that another thread can come along and mess with the wait_set (possibly removing entities), which means that the rest of the code could be accessing memory that has already been freed.

To test this, I ended up commenting out the release of the GIL above, and with that in place, I can't reproduce the crash anymore.  That said, that is not a proper solution so I'm going to have to look into adding more fine-grained locking into rclpy and see if I can fix the issue that way.  More to come on this later.

clalancette, 2022-03-10:
I do believe that this was fixed by ros2/rclpy#904, so closing this out.  Feel free to reopen if I'm wrong.

PR #504 (2022-02-17): Add in one more destroy call that was missed in testing.
This may fix some of the crashes on Windows.  Either way,
it doesn't hurt to have (and is more "correct" in our
current usage of rclpy).

This is a follow-on from #499 , and may fix #503