ros2/geometry2 #519: Can't build tf2_geometry_msgs: "fatal error: kdl/frames.hpp: No such file or directory"

Kaju-Bubanja, 2022-04-19:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - master(from source)
- Version or commit hash:
  - ef2dd147a25f14ed946c86f5a9fc03c8995d8fbe
- DDS implementation:
  - rmw_fastrtps_cpp
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue
Trying to build tf2_geometry_msgs leads to following error
```
colcon build --symlink-install --packages-up-to tf2_geometry_msgs
Starting >>> ament_package
Starting >>> ament_lint
Starting >>> ament_cppcheck
Starting >>> gtest_vendor
Starting >>> fastcdr
Starting >>> osrf_testing_tools_cpp
Starting >>> google_benchmark_vendor
Starting >>> osrf_pycommon
Finished <<< fastcdr [0.71s]                                                                                                                                                                                      
Starting >>> ament_pycodestyle
Finished <<< google_benchmark_vendor [0.78s]                                                                                                                                    
Finished <<< gtest_vendor [0.97s]                                                                                                                                                                      
Starting >>> gmock_vendor                                                                                                                                                                       
Finished <<< osrf_testing_tools_cpp [1.05s]
Finished <<< gmock_vendor [0.17s]                                                                                                                                                     
Finished <<< ament_package [1.34s]                                                                                                                      
Starting >>> ament_cmake_core
Finished <<< osrf_pycommon [1.35s]
Finished <<< ament_lint [1.37s]
Starting >>> ament_flake8
Finished <<< ament_cppcheck [1.38s]                                                                                                         
Finished <<< ament_cmake_core [0.25s]                                                                                 
Starting >>> ament_cmake_libraries
Starting >>> ament_cmake_export_libraries
Starting >>> ament_cmake_python
Starting >>> ament_cmake_include_directories
Starting >>> ament_cmake_export_definitions
Starting >>> ament_cmake_export_include_directories
Finished <<< ament_pycodestyle [0.95s]
Starting >>> ament_cmake_export_link_flags                                                                                                                                       
Finished <<< ament_cmake_libraries [0.20s]                                                                                                                                             
Starting >>> ament_cmake_export_dependencies
Finished <<< ament_cmake_python [0.23s]
Finished <<< ament_cmake_include_directories [0.23s]
Starting >>> ament_cmake_test
Finished <<< ament_cmake_export_definitions [0.24s]
Starting >>> ament_cmake_version
Starting >>> ament_cmake_target_dependencies
Finished <<< ament_cmake_export_libraries [0.28s]
Finished <<< ament_cmake_export_include_directories [0.26s]
Starting >>> ament_cmake_export_targets                                                                                                                                               
Starting >>> ament_cmake_export_interfaces
Finished <<< ament_cmake_export_link_flags [0.24s]
Starting >>> test_interface_files
Finished <<< ament_cmake_export_dependencies [0.25s]                                                                                                                                
Finished <<< ament_cmake_version [0.21s]
Finished <<< ament_cmake_target_dependencies [0.21s]
Finished <<< ament_cmake_export_targets [0.23s]                                                                                                                                          
Finished <<< ament_cmake_export_interfaces [0.23s]
Finished <<< test_interface_files [0.19s]
Finished <<< ament_flake8 [0.86s]                                                           
[... 215 lines cut ...]
Starting >>> tf2_geometry_msgs
--- stderr: tf2_geometry_msgs                                  
In file included from /home/user/ros2_rolling/src/ros2/geometry2/tf2_geometry_msgs/test/test_tf2_geometry_msgs.cpp:45:
/home/user/ros2_rolling/src/ros2/geometry2/tf2_geometry_msgs/include/tf2_geometry_msgs/tf2_geometry_msgs.hpp:51:10: fatal error: kdl/frames.hpp: No such file or directory
   51 | #include "kdl/frames.hpp"
      |          ^~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/test_tf2_geometry_msgs.dir/build.make:63: CMakeFiles/test_tf2_geometry_msgs.dir/test/test_tf2_geometry_msgs.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:184: CMakeFiles/test_tf2_geometry_msgs.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [Makefile:141: all] Error 2
---
Failed   <<< tf2_geometry_msgs [0.52s, exited with code 2]

Summary: 133 packages finished [23.0s]
  1 package failed: tf2_geometry_msgs
  1 package had stderr output: tf2_geometry_msgs

```

#### Expected behavior
No compilation error
#### Actual behavior
See error above
#### Additional information

clalancette, 2022-04-19:
It looks like you are probably missing `orocos_kdl_vendor` from your installation, which `tf2_geometry_msgs` requires.  I would suggest doing the following in the root of your workspace:

```
$ rm -f ros2.repos
$ wget https://raw.githubusercontent.com/ros2/ros2/master/ros2.repos
$ vcs import src < ros2.repos
$ vcs pull src
```

And then run your compile again.

Kaju-Bubanja, 2022-04-19:
What seems odd to me is that in the package.xml it says `<depend>orocos_kdl_vendor</depend>` but in the CMakeLists.txt it says `find_package(orocos_kdl REQUIRED)` usually these two names are the same. Is that deliberate here?