ros2/geometry2 #527: How to get static transform once?

TemugeB, 2022-05-26:
## Bug report

**Required Info:**

- Operating System:
Ubuntu 20.04
- Installation type:
apt install ros-foxy-desktop
- Version or commit hash:
geometry2: 0.13.13
- DDS implementation:
  - 
- Client library (if applicable):
rclpy

#### Steps to reproduce issue

Run dummy transformation
```
ros2 run tf2_ros static_transform_publisher 0 0 0 0 0 0 workspace camera_link
```

Try to read the static transform:
```python
import rclpy 
from tf2_ros.buffer import Buffer
from tf2_ros.transform_listener import TransformListener
from rclpy.node import Node
from rclpy.duration import Duration

class Plane(Node):

    def __init__(self):

        super().__init__('plane')

        self.tf_buffer = Buffer(cache_time=Duration(seconds = 10.))
        self.tf_listener = TransformListener(self.tf_buffer, self)

    def get_transform(self):
        
        now = rclpy.time.Time()

        try:
            trans = self.tf_buffer.lookup_transform('camera_link', 'camera_link', now, timeout=Duration(seconds=2.))
            print(trans)
        except Exception as e:
            print('camera_link camera_link lookup failed')

        try:
            self.tf_buffer.lookup_transform('camera_link', 'workspace', now, timeout=Duration(seconds=2.))
            print(trans)
        except Exception as e:
            print('camera_link workspace lookup failed: ' + str(e))

        try:
            self.tf_buffer.lookup_transform('workspace', 'workspace', now, timeout=Duration(seconds=2.))
            print(trans)
        except Exception as e:
            print('workspace workspace lookup failed')

def main():
    rclpy.init()
    plane = Plane()
    plane.get_transform()

if __name__ == '__main__':
    main()
```

#### Expected behavior

Static transform should be obtained.

#### Actual behavior

"camera_link" "workspace" transformation lookup fails with the following error.
```
"camera_link" passed to lookupTransform argument target_frame does not exist.
```
camera_link--camera_link and workspace--workspace transforms are returned fine. 

This
```
ros2 run tf2_ros tf2_echo camera_link workspace
```
produces this output:
```
[INFO] [1653543600.123318542] [tf2_echo]: Waiting for transform camera_link ->  workspace: Invalid frame ID "camera_link" passed to canTransform argument target_frame - frame does not exist
At time 0.0
- Translation: [0.000, 0.000, 0.000]
- Rotation: in Quaternion [0.000, 0.000, 0.000, 1.000]
At time 0.0
- Translation: [0.000, 0.000, 0.000]
- Rotation: in Quaternion [0.000, 0.000, 0.000, 1.000]
```

So ```tf2_echo``` also seems to have difficulty finding the frame initially but eventually returns the transformation.

#### Additional information

I'm not sure if I'm doing something wrong or if I am facing a bug. The TF2 tutorial shows how to read transforms using a timer callback. I just need to get the transform once at the beginning of my program between the workspace and a fixed camera. But I can't find any examples on how to read static transforms. Am I missing something? Any help would be greatly appreciated.

----
## Feature request

#### Feature description

#### Implementation considerations

TemugeB, 2022-05-26:
This is my work around. Is this the correct way to do this?

```python
import rclpy 
from tf2_ros.buffer import Buffer
from tf2_ros.transform_listener import TransformListener
from rclpy.node import Node
from rclpy.duration import Duration

class Plane(Node):

    def __init__(self):

        super().__init__('plane')

        self.tf_buffer = Buffer(cache_time=Duration(seconds = 10.))
        self.tf_listener = TransformListener(self.tf_buffer, self)
        self.timer = self.create_timer(1.0, self.get_transform)

    def get_transform(self):
        
        try:
            now = rclpy.time.Time()
            trans = self.tf_buffer.lookup_transform('camera_link', 'workspace', now, timeout=Duration(seconds=1.0))
        except Exception as e:
            print('could not find the transformation: ' + str(e))
        else:
            self.trans = trans
            self.timer.cancel()
            print(self.trans)

def main():
    rclpy.init()
    plane = Plane()
    plane.get_transform()

if __name__ == '__main__':
    main()
```

TemugeB, 2022-05-26:
My work around is to use timer and exit it once the transformation is found. But is this the correct way?
```python
import rclpy 
from tf2_ros.buffer import Buffer
from tf2_ros.transform_listener import TransformListener
from tf2_ros import TransformException
from rclpy.node import Node
from rclpy.duration import Duration

class Plane(Node):

    def __init__(self):

        super().__init__('plane')

        self.tf_buffer = Buffer(cache_time=Duration(seconds = 10.))
        self.tf_listener = TransformListener(self.tf_buffer, self)
        self.timer = self.create_timer(1.0, self.get_transform)
        self.trans = None

    def get_transform(self):
        
        try:
            now = rclpy.time.Time()
            trans = self.tf_buffer.lookup_transform('camera_link', 'workspace', now, timeout=Duration(seconds=1.0))
            self.trans = trans
            print(trans)
            self.timer.cancel()
        except TransformException as e:
            print('could not find the transformation: ' + str(e))

def main():
    rclpy.init()
    plane = Plane()
    while rclpy.ok() and plane.trans == None:
        rclpy.spin_once(plane)

if __name__ == '__main__':
    main()
```

clalancette, 2022-05-27:
For these kinds of questions, please ask them on https://answers.ros.org.  That is our central question and answers site, and will ensure that the answer is searchable in the future.

I'm going to close this one out for now.

closed 2022-05-27 by clalancette

TemugeB, 2022-05-30:
If anyone comes here with the same question, this is what I ended up doing.
```python 
import rclpy 
from tf2_ros.buffer import Buffer
from tf2_ros.transform_listener import TransformListener
from tf2_ros import TransformException
from rclpy.node import Node
from rclpy.duration import Duration

class Plane(Node):

    def __init__(self):

        super().__init__('plane')

        self.tf_buffer = Buffer(cache_time=Duration(seconds = 10.))
        self.tf_listener = TransformListener(self.tf_buffer, self)
        self.timer = self.create_timer(.1, self.get_transform)
        self.trans = None

        while rclpy.ok() and self.trans == None:
            rclpy.spin_once(self)

        self.destroy_node()

    def get_transform(self):
        
        try:
            now = rclpy.time.Time()
            trans = self.tf_buffer.lookup_transform('camera_link', 'workspace', now, timeout=Duration(seconds=1.0))
            self.trans = trans
            print(trans)
            self.timer.cancel()
        except TransformException as e:
            print('could not find the transformation: ' + str(e))

def main():
    rclpy.init()
    plane = Plane()

if __name__ == '__main__':
    main()
```

```timer.cancel()``` call is likely not necessary since ```destroy_node()``` automatically destroys timers.