ros2/geometry2 #550: Documenting CLI tools?

zflat, 2022-08-11:
Is there any documentation of `tf_echo` and other tools that can be run from the command line that is equivalent to http://wiki.ros.org/tf/Debugging%20tools and http://wiki.ros.org/tf2_ros#Tools ?

I think that http://docs.ros.org/en/humble/p/tf2_ros/ would be the correct place for such documentation.

clalancette, 2022-08-11:
Agreed that that would be the correct place.  Right now, there is no documentation like that, but we'd be happy to review pull requests adding it.

CursedRock17, 2024-02-16:
I would like to pick this up if possible, but to have a more centralized documentation, should we create a https://docs.ros.org/en/humble/p/geometry2/ ? Then, we can take our examples and put them underneath an examples section within the tf2_ros section of the documentation.

clalancette, 2024-02-16:
> I would like to pick this up if possible, but to have a more centralized documentation, should we create a https://docs.ros.org/en/humble/p/geometry2/ ? Then, we can take our examples and put them underneath an examples section within the tf2_ros section of the documentation.

So the way that this works is that the documentation is auto-generated by the buildfarm.  But it is uploaded on a per-package, not a per-repository basis.  Thus we already have the correct directory: http://docs.ros.org/en/humble/p/tf2_ros/ .

That said, our usual practice is to make changes to `rolling`, and then backport as needed.  So I'll suggest starting there.