ros2/geometry2 #554: TransformBroadcaster takes shared pointer or raw node?

ijnek, 2022-09-20:
## Bug report

**Required Info:**

- Operating System: Ubuntu 22.04 Jammy
- Installation type: Source
- Version or commit hash: rolling
- DDS implementation: Rolling Default

### Description

The [TransformBroadcaster constructor](https://github.com/ros2/geometry2/blob/d6559f955c3f9de3360d35e10373d7c44185db4a/tf2_ros/include/tf2_ros/transform_broadcaster.h#L59) should take a ``NodeT``, but tests in [tf2_ros/test/test_transform_broadcaster.cpp](https://github.com/ros2/geometry2/blob/d6559f955c3f9de3360d35e10373d7c44185db4a/tf2_ros/test/test_transform_broadcaster.cpp) pass a pointer to a rclcpp::Node to the broadcaster (and the tests pass although they shouldn't).

Changes in https://github.com/ros2/geometry2/pull/552 exposed this issue, as it depends on ``NodeT`` being a pointer to a rclcpp::Node.

In [Writing a broadcaster (C++)](https://docs.ros.org/en/rolling/Tutorials/Intermediate/Tf2/Writing-A-Tf2-Broadcaster-Cpp.html), a node instance is passed to the constructor as below:

```cpp
tf_broadcaster_ = std::make_unique<tf2_ros::TransformBroadcaster>(*this);
```

But in the [tf2_ros/test/test_transform_broadcaster.cpp](https://github.com/ros2/geometry2/blob/d6559f955c3f9de3360d35e10373d7c44185db4a/tf2_ros/test/test_transform_broadcaster.cpp#L53), a shared pointer is provided.

```cpp
TEST(tf2_test_transform_broadcaster, transform_broadcaster_rclcpp_node)
{
  auto node = rclcpp::Node::make_shared("tf2_ros_message_filter");

  tf2_ros::TransformBroadcaster tfb(node);
}
```

This was originally reported in https://github.com/ros-perception/image_pipeline/issues/775.

#### Steps to reproduce issue

```
mkdir -p geometry2_ws/src
cd "$_"
git clone git@github.com:ijnek/geometry2 -b ijnek-bug-test-case
cd ../
colcon build
```

#### Expected behavior

Compiles fine.

#### Actual behavior

Doesn't compile.

```
--- stderr: tf2_ros                               
In file included from /home/ijnek/tmp_workspaces/geometry2_ws/src/geometry2/tf2_ros/test/test_transform_broadcaster.cpp:31:
/home/ijnek/tmp_workspaces/geometry2_ws/src/geometry2/tf2_ros/include/tf2_ros/transform_broadcaster.h: In instantiation of ‘tf2_ros::TransformBroadcaster::TransformBroadcaster(NodeT&&, const rclcpp::QoS&, const rclcpp::PublisherOptionsWithAllocator<AllocatorT>&) [with NodeT = rclcpp::Node&; AllocatorT = std::allocator<void>]’:
/home/ijnek/tmp_workspaces/geometry2_ws/src/geometry2/tf2_ros/test/test_transform_broadcaster.cpp:57:41:   required from here
/home/ijnek/tmp_workspaces/geometry2_ws/src/geometry2/tf2_ros/include/tf2_ros/transform_broadcaster.h:71:7: error: base operand of ‘->’ has non-pointer type ‘rclcpp::Node’
   71 |       node->get_node_parameters_interface(),
      |       ^~~~
gmake[2]: *** [CMakeFiles/tf2_ros_test_transform_broadcaster.dir/build.make:76: CMakeFiles/tf2_ros_test_transform_broadcaster.dir/test/test_transform_broadcaster.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:510: CMakeFiles/tf2_ros_test_transform_broadcaster.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< tf2_ros [14.3s, exited with code 2]
```

alsora, 2022-09-21:
Yes, the changes in https://github.com/ros2/geometry2/pull/552 require that the `NodeT` passed to the constructor of `TransformBroadcaster` is some sort of pointer.

Note that this requirement was already present in the constructor of `TransformListener`: the `node` passed to the constructor was then used in the `init` function as `node->get_node_base_interface()` (thus assuming that it was a pointer) see https://github.com/ros2/geometry2/pull/552/files#diff-db2161f65f99c88d9aef0e89c8d8b2774fcc52e45c9ccc47a9ab76648842f908L102 and https://github.com/ros2/geometry2/pull/552/files#diff-db2161f65f99c88d9aef0e89c8d8b2774fcc52e45c9ccc47a9ab76648842f908L119

Before my PR `TransformBroadcaster` was accepting either nodes or pointers to nodes, while `TransformListener` was accepting only pointers to nodes.
Now they both only accept pointers to nodes.
I wasn't aware of the existence of that tutorial where a node was used.

There are different options here:
 1. change the tutorial to use a pointer.
 2. change the tutorial to use a pointer and also add a deprecated constructor that takes a node to preserve backward compatibility.
 3. change both `TransformBroadcaster` and `TransformListener` to accept both nodes and pointers to nodes.

IMO nodes should handled via pointers to make it harder to do mistakes, thus my preference would be (2).

ijnek, 2022-09-21:
I agree with option 2 too.

Could we change [this line](https://github.com/ros2/geometry2/blob/d6559f955c3f9de3360d35e10373d7c44185db4a/tf2_ros/include/tf2_ros/transform_broadcaster.h#L59) from

For the new method, instead of having:

```cpp
  template<class NodeT, class AllocatorT = std::allocator<void>>
  TransformBroadcaster(
    NodeT && node
```

could I suggest having:

```cpp
template<class NodeT = rclcpp::Node, class AllocatorT = std::allocator<void>>
  TransformBroadcaster(
    std::shared_ptr<NodeT> node,
....
```

That's how its done in a [function from rclcpp/executors.hpp](https://github.com/ros2/rclcpp/blob/33dae5d679751b603205008fcb31755986bcee1c/rclcpp/include/rclcpp/executors.hpp#L85).

clalancette, 2022-09-21:
> I agree with option 2 too.

Yes, I agree with this too.  @alsora could you make the updates?  Thanks.

alsora, 2022-09-21:
Yes, will open follow-up PRs soon.

clalancette, 2022-09-21:
> Yes, will open follow-up PRs soon.

Thanks, appreciated.

ijnek, 2022-09-23:
> Yes, will open follow-up PRs soon. @ijnek note that your proposed approach would not work with raw pointers.

I was just trying to get across the point that NodeT itself shouldn't be a pointer, but should be the rclcpp node type itself. That's how it is everywhere else in rclcpp.

In regards to using shared_ptr vs raw pointer vs reference to pass rclcpp::Node, I don't know what the recommended way is.

closed 2022-09-29 by clalancette