ros2/geometry2 #561: Compile error while building `tf2_kdl` with `BUILD_TESTING` on

pijaro, 2022-11-08:
## Bug report

**Required Info:**

- Operating System:
  - Windows 10
- Installation type:
  - source, with `BUILD_TESTING`
- Version or commit hash:
  - ae2120833d1777eaa91a3fa4943d56b675db3e52 (humble branch)
- DDS implementation:
  - fastrtps
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue
Build `geometry2` package with `BUILD_TESTING` on.

```
colcon build --cmake-args -DBUILD_TESTING:int=1
```

#### Expected behavior
Build succeeded

#### Actual behavior
Compile error for `tf2_kdl` package:
```
D:\ros2-humble\include\orocos_kdl\kdl\jacobian.hpp(26,10): fatal error C1083: cannot open include file 'Eigen/Core' no such file or directory
```

#### Additional information
Looks like Eigen is not added to include dirs. Adding it fixes the issue:

```
if(BUILD_TESTING)
  find_package(ament_cmake_gtest REQUIRED)
  find_package(rclcpp REQUIRED)
  find_package(tf2_msgs REQUIRED)
  find_package(Eigen3 REQUIRED)

  ament_add_gtest(test_kdl test/test_tf2_kdl.cpp)
  target_include_directories(
    test_kdl PRIVATE
    "${Eigen3_INCLUDE_DIRS}"
  )
  target_link_libraries(test_kdl
    tf2_kdl
    rclcpp::rclcpp
    # Used, but not linked to test tf2_kdl's exports:
    #   tf2::tf2
    #   tf2_ros::tf2_ros
  )
endif()
```

clalancette, 2022-11-08:
The thing is that the error is coming from orocos_kdl, not from geometry2.  So it seems like a problem in orocos_kdl (or orocos_kdl_vendor), not in geometry2.

That said, we do compile and run these tests nightly on Windows, and we haven't seen the same problem (see https://ci.ros2.org/view/nightly/job/nightly_win_deb/2538/console , for instance).  So I'm guessing that somehow your installation differs from ours somehow.  How did you install Eigen and orocos_kdl?

pijaro, 2022-11-09:
I'm using the binary release of ROS2 humble for windows ([link](https://github.com/ros2/ros2/releases/tag/release-humble-20220523)), so I believe orocos_kdl comes with it. 

As for Eigen, I followed ROS2 install instructions and installed it via choco [(link)](https://docs.ros.org/en/humble/Installation/Windows-Install-Binary.html#install-dependencies).

pijaro, 2022-11-09:
I'm not a CMake expert, but I think the Eigen from choco (3.3.4) doesn't have exported targets. Therefore it might be the reason why it doesn't work.

The "C:\ProgramData\chocolatey\lib\eigen\share\cmake" directory contains only:
- `Eigen3Config.cmake`, and
- `Eigen3ConfigVersion.cmake`

I see there are no `*Targets.cmake` or anything similar.

clalancette, 2022-11-09:
Yes, that is definitely true, and a longstanding issue that we haven't gotten around to fixing.  But the thing is that we build these tests nightly, so I'm still a bit confused on why it didn't work for you.  Probably needs a deeper look into the exports and the problem to figure out exactly what is going on.

mjs513, 2023-01-03:
Hi 
I am having he same issue with the exception I downloaded eigen3 and building it along with geometry2 in a ros2 workspace.
[image]

I installed ros2 humble for windows 10 using the binary and orocos_kdl is installed as part of humble.  also installed the packages the same way with choco from here: https://docs.ros.org/en/humble/Installation/Windows-Install-Binary.html#install-dependencies
```
Please download these packages from [this](https://github.com/ros2/choco-packages/releases/latest) GitHub repository.

    asio.1.12.1.nupkg
    bullet.3.17.nupkg
    cunit.2.1.3.nupkg
    eigen-3.3.4.nupkg
    tinyxml-usestl.2.6.2.nupkg
    tinyxml2.6.0.0.nupkg

Once these packages are downloaded, open an administrative shell and execute the following command:

choco install -y -s <PATH\TO\DOWNLOADS\> asio cunit eigen tinyxml-usestl tinyxml2 bullet
```

But no matter what I try I keep getting the same error:
`C:\dev\ros2_humble\include\orocos_kdl\kdl\jacobian.hpp(26,10): fatal error C1083: Cannot open include file: 'Eigen/Core': No such file or directory [C:\dev\ros2_ws\build\tf2_kdl\test_kdl.vcxproj]`

I did find this from 2013, https://www.orocos.org/forum/orocos/orocos-users/eigen3-include-problem-kdl, I did check that findeigen3.cmake is in kdl and a couple of other places.