ros2/geometry2 #567: Derive ExtrapolationException for "too in the past/future" cases

roncapat, 2022-12-01:
Hello, I would like to propose an extension to the current exceptions set.
In particular, tf2::ExtrapolationException does not specify whether the "Lookup would require extrapolation into the past" or "Lookup would require extrapolation into the future". It may be wise to distinguish the two cases: eg. for the "past" case, it may be unwise to call lookupTransform() again since it may be unlikely to receive such an old-stamped needed transform on the TF2 broadcast system, so the user may want to, for example, abort the current data processing and skip over to the next message. For the future case, it may be useful instead to retry the lookupTransform() call.
Even if the propose use case are not solid enought, providing tf2::BackwardExtrapolationException and tf2::ForwardExtrapolationException as classes derived from tf2::ExtrapolationException may be informative in some way to the user (instead of parsing the .what() string for past/ future keywords.

Any opinions / would a PR be considered?

tfoote, 2022-12-15:
It would be fine to add those sub classes. There's no specific reason not to capture that. We stopped there to keep the complexity down. But this is stable enough that we can improve the documentation and tutorials and make sure to extend coverage of these cases to throw them separately.

roncapat, 2022-12-15:
For a minimal C++ patch, it's just a matter of editing the following three files:

https://github.com/ros2/geometry2/blob/599fa7c5dd050fd0978d7b1b4a9fca0ebb60410b/tf2/include/tf2/exceptions.h#L140-L148

https://github.com/ros2/geometry2/blob/599fa7c5dd050fd0978d7b1b4a9fca0ebb60410b/tf2_ros/src/buffer_client.cpp#L160-L162

https://github.com/ros2/geometry2/blob/599fa7c5dd050fd0978d7b1b4a9fca0ebb60410b/tf2_ros/src/buffer_server.cpp#L78-L81

+ check and fix all occurrences where now ExtrapolationException appears in documentation strings.

Then, propagation to Python.

https://github.com/ros2/geometry2/blob/599fa7c5dd050fd0978d7b1b4a9fca0ebb60410b/tf2_py/src/tf2_py.cpp#L74-L78

https://github.com/ros2/geometry2/blob/599fa7c5dd050fd0978d7b1b4a9fca0ebb60410b/tf2_ros_py/tf2_ros/buffer_client.py#L267

If you want, I can work for a PR with a first C++ [patch 1] + Python [patch 2] implementation.
Final documentation touches in a 3rd patch.

Existing code and tutorials catching simply ExtrapolationException would continue to be valid, but in future work should be extended too (I do not searched for a full list of tutorials to update yet).

roncapat, 2022-12-17:
It might be more complex... in fact, digging more, I found out that TimeCacheInterface::getData() only returns false on unavailable data, and fills a string with one of three possible error messages (like: can't extrapolate, too in the past, to in the future). That is an API bottleneck that does not allow BufferCore::walkToTopParent to detect which Exception to throw without analyzing the error string filled by TimeCache.

So: TimeCacheInterface API needs to be broken (or extended). In your opinion, apart from packages inside the geometry2 package, is there any potential that someone uses directly a TimeCache from tf2 package? Do you have any ideas on how to modify the API? Personally, I would return int instead of bool from TimeCacheInterface::getData to allow for differentiation of returned error condition.

tfoote, 2023-02-11:
The TimeCacheInterface is basically internal to the tf2 API and we could extend it for release in a new distribution.

roncapat, 2023-02-17:
@tfoote I made a try with the linked PR (#586).

closed 2023-03-23 by roncapat