ros2/geometry2 #570: Feelings about using ROS2 Logging?

Aposhian, 2022-12-05:
## Feature request

#### Feature description
Use `RCLCPP_LOG_*` or `RCUTILS_LOG_*` instead of [console_bridge](https://github.com/ros/console_bridge), which hasn't been updated for years. `console_bridge` simply adds additional boilerplate to be able to unify `tf2` logs with the ROS 2 components in which it is usually embedded (example: https://github.com/ros/rosconsole_bridge/issues/21).

Even if you implement an output handler for `console_bridge`, you still don't get things like the function name.

`console_bridge` is designed to be "ROS agnostic", but seems that `tf2` is primarily a ROS 2 package. If we want to minimize the ROS 2 dependency, we could even make it use `rcutils` instead of `rclcpp`.

CursedRock17, 2024-03-08:
I believe this issue can be closed  if the only intentions were to get rid of `console_bridge` but I'm going to open issues/PRs for `urdfdom` and `class_loader`

mjcarroll, 2024-03-08:
Feel free to open corresponding tickets on those repos.  Thanks for iterating!

closed 2024-03-08 by clalancette