ros2/geometry2 #585: rclpy  StaticTransformBroadcaster does not keep all transforms for new new subscribers (topic not transient local?)

firesurfer, 2023-02-15:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - binaries
- Version or commit hash:
  - humbly
- DDS implementation:
  - cyclone
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue

I ran into the issue that only one transform published with a     `StaticTransformBroadcaster` will be recieved by any `TransformListeners` that are started afterwards.

In order to reproduce simply run a robot_state_publisher with a rather complex robot model (like 100 links or so) and have an additional node publish some static_tfs

For copy paste
```
import rclpy
from rclpy.node import Node
from tf2_geometry_msgs.tf2_geometry_msgs import  TransformStamped
from tf2_ros.static_transform_broadcaster import StaticTransformBroadcaster

def main(args=None):
    rclpy.init(args=args)
    node = Node("Test_tf")

    broadcaster = StaticTransformBroadcaster(node)

    for i in range(0, 100):
        tf = TransformStamped()
        tf.child_frame_id = "abc" + str(i)
        tf.header.frame_id = "hallo"

        broadcaster.sendTransform(tf)

    rclpy.spin(node)

if __name__ == "__main__":
    main()

``

Then run 
`ros2 topic echo /tf_static`

You will notice that some static transforms will be missing.

### Background

If one takes a look at the source of the `StaticTransformBroadcaster` one notices that the queue depth is 1
```
qos = QoSProfile(
                depth=1,
                durability=DurabilityPolicy.TRANSIENT_LOCAL,
                history=HistoryPolicy.KEEP_LAST,
                )
```

So it makes sense that only the last transform is published.

### Issue

This is rather counter intuitive. If I use a StaticTransformBroadcaster I expect it to keep all transforms that I gave available for new subscribers.

clalancette, 2023-02-15:
> This is rather counter intuitive. If I use a StaticTransformBroadcaster I expect it to keep all transforms that I gave available for new subscribers.

Can you explain why this is counter-intuitive?  It is a static transform, so only the last one should matter.

firesurfer, 2023-02-15:
From my understanding I can give a StaticTransformBroadcaster a transform, and then another transform when ever I will and as long as a new transform does not have the same links involved it is simply another transform I expect it publish.

From my point of view both transforms are valid, independent of the order I pass them to the broadcaster

clalancette, 2023-02-15:
> From my point of view both transforms are valid, independent of the order I pass them to the broadcaster

But if you are changing the transforms, why are you publishing them statically?  They should just be regular transforms then.

firesurfer, 2023-02-16:
What I meant:

I have a program with one StaticTransformBroadcaster and I would like to use that to publish at different time points static transforms (Each transform will not change over time, but over time I want to add new static transforms)

At the moment this is only possible by either storing a list of all transforms in my program and publish them or by creating new broadcasters

firesurfer, 2023-02-23:
But yeah probably just a different point of view / expectations on how things should behave. Closing this issue therefore.

closed 2023-02-23 by firesurfer