ros2/geometry2 #589: [bug

Aposhian, 2023-03-15:
## Bug report

**Required Info:**

- Operating System:
  - 
- Installation type:
  - 
- Version or commit hash:
  - 
- DDS implementation:
  - 
- Client library (if applicable):
  - 

#### Steps to reproduce issue

```

```

#### Expected behavior

#### Actual behavior

#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

clalancette, 2023-03-23:
This seems like a reasonable feature request.  We'd be happy to review a PR that implements this.

vineet131, 2023-03-28:
> You will want to switch the order of those checks, since `checkAndErrorDedicatedThreadPresent` logs if it fails the check. If you check for timeout first then it will shortcut and not evaluate the check.

You're right about that, that's my bad. I preferred approach 2 to approach 1 because those who have implemented this code should be able to transform as is without having to pass another kind of signature. I'll write up a PR for this.

peci1, 2025-04-09:
This can be closed as #593 fixed it.

closed 2025-04-09 by christophebedard