ros2/geometry2 #600: #586 done broke me

SteveMacenski, 2023-05-04:
A full thread can be found https://github.com/osrf/ros2_test_cases/issues/1042 but during Iron Testing Day, it was identified that Nav2 is totally busted and jumping around. Upon further investigation it was isolated to `geometry2`, upon further investigation again it was found to be the cause of 2 commit
- https://github.com/ros2/geometry2/commit/3bac620777043a1c8761cd74c1bb814667341b65
- https://github.com/ros2/geometry2/commit/7afeb6c2b300be5a6b97eed0fa822a1d294e7b77

 I cannot tell you why in particular these are broken, but they should either be reverted for Iron or the issue resolved before release. Until such a time, the Nav2 stack does some wacky stuff that its hard to indicate to you clearly how these commits are causing it. 

My best description of what I see:

> We're seeing Nav2 receiving AMCL localization service requests, but not honoring the content of them (and seems to randomly just bounce around) & Gazebo Classic appears to be producing data that doesn't seems sensible for where the robot truly is in the environment + reporting that the robot is outside of the local costmap bounds for update.

roncapat, 2023-05-04:
Hi Steve, I'm sorry for that. But I'm a bit confused about why this can happen at all.

Those changes deal only with the exception paths of TF2, so it might be needed to understand if/how/where those exceptions are handled by Nav2 in the case you mentioned.

Could you please share/point the standard nav2 configuration launch scenario you mentioned, so that I can have a look? I use Nav2 with a different setup, no AMCL for example.

Does Nav2 breaks also in other test cases/configurations?

clalancette, 2023-05-04:
Yes, I have the same questions as @roncapat ; these changes seem innocuous, so I'm surprised they are the cause of the issue.  Having a way to reproduce so we can look into the problem would be most helpful.

I'd prefer to try to debug it and fix it forward first.  If we can't make progress there, then I'm not opposed to reverting #586 and #592.

SteveMacenski, 2023-05-04:
I'll try a few things.  I can tell you its not https://github.com/ros2/geometry2/pull/592/files since its broken before it. So I'm going to make a baseline guess that its nothing to do with the error codes themselves or that setting of values.

The 2 things that draw my eye are

```
      return tf2::TF2Error::TF2_EXTRAPOLATION_ERROR;
      return error_code;
```

and the new exceptions

```
      case tf2::TF2Error::TF2_BACKWARD_EXTRAPOLATION_ERROR:
        throw BackwardExtrapolationException(error_string);
      case tf2::TF2Error::TF2_FORWARD_EXTRAPOLATION_ERROR:
        throw ForwardExtrapolationException(error_string);
      case tf2::TF2Error::TF2_NO_DATA_FOR_EXTRAPOLATION_ERROR:
        throw NoDataForExtrapolationException(error_string);
```

My speculation is that one of two things are happening: (1) something consuming exceptions further up the stack just haven't been programmed to use them so they're not being handled properly (2) the error code setting isn't working properly, so `return error_code` is just garbage in one or more places.

SteveMacenski, 2023-05-04:
Fun stuff, found it, I was correct its 

```
-      return tf2::TF2Error::TF2_EXTRAPOLATION_ERROR;
+      return error_code;
```

Reverting those 2 instances, things work again. Something's up with populating `error_code`

roncapat, 2023-05-04:
I'm in... Tomorrow evening I'll give an in depth look, but of course everything you can find or narrow down it's a nice to have (while I was writing, I already saw you found something, so thank you!).

SteveMacenski, 2023-05-04:
Its actually just one of those lines, L461 changing it back fixes the problem. I spent about 3 minutes trying to understand why, and I didn't come up with an answer, but that's the targeted problem.

roncapat, 2023-05-04:
Now I'm from mobile but... I noticed that on line 

https://github.com/ros2/geometry2/blob/rolling/tf2/src/cache.cpp#L124

I set an error but not an error string. A bit over the lines you reverted, there is an "if(error_string) to detect the presence of an error.

SteveMacenski, 2023-05-04:
Maybe this is an easy fix then!

roncapat, 2023-05-06:
I'm 90% sure I found it in code. There's one possible overwrite/reset of error_code... Will test and post the patch soon hopefully.

roncapat, 2023-05-06:
@SteveMacenski coud you try https://github.com/roncapat/geometry2? I posted a patch... the rationale is that we have `extrapolation_error_string` potentially holding a different message than `error_string`. So same must be for `error_code` (but wasn't). 

In principle, since before the introduction of `error_code` in the API the only thing carried around was the string (describing an error only in a lexical way), now it's just a matter of bringing around together both string and code representation of the error being careful not to mix/decouple them.

SteveMacenski, 2023-05-06:
That fixes it! Can you submit the PR?

@clalancette what's the procedure for getting a patch into the distribution during testing (anything I can / should do)?

clalancette, 2023-05-07:
> @clalancette what's the procedure for getting a patch into the distribution during testing (anything I can / should do)?

CI needs to run, then we need to backport to Iron, and then we need to do patch releases.  Not much to do here, we should be able to get this all in the next couple of days.

closed 2023-05-08 by clalancette