ros2/geometry2 #606: Python tf2 buffer does not Detect Jumps in Time

erichlf, 2023-05-31:
In `humble`, after setting `use_sim_time` and running a bag in a loop via
`ros2 bag play -l --clock 10 <bag>`
I get the `Warning: TF_OLD_DATA ignoring data from the past for frame tilt at time ...` error message. This does not occur when using `rclcpp` and only `rclpy`. I have verified that in the source code for `buffer.py` the logic for detecting clock jumps is missing, but is in the source for `buffer.cpp`. Minimum code to reproduce is as follows:
```
#!/usr/bin/env python3

import rclpy
from rclpy.node import Node
from rclpy.qos import qos_profile_sensor_data
from sensor_msgs.msg import PointCloud2
from tf2_ros import TransformException
from tf2_ros.buffer import Buffer
from tf2_ros.transform_listener import TransformListener

class TestNode(Node):
    def __init__(self):
        super().__init__("test_node")
        # Subscribe to /points
        self.get_logger().info("Subscribing to PointCloud2 topic...")
        self.subscription = self.create_subscription(
            PointCloud2,
            "/points",
            self.callback_main,
            qos_profile_sensor_data,
        )
        self.tf_buffer = Buffer()
        self.tf_listener = TransformListener(self.tf_buffer, self)

    def callback_main(self, pc_msg):
        self.get_logger().info(f"time = {self.get_clock().now()}")

def main(args=None):
    # Init client library
    rclpy.init(args=args)
    # Create the pub/sub
    test_node = TestNode()
    # Spinning...
    try:
        rclpy.spin(
            test_node,
        )
    except KeyboardInterrupt:
        pass

    return 0
```

And I ran this simply by
`ros2 run test test_node --ros-args -p use_sim_time:=true`

Hermanoid, 2023-06-13:
I also had this problem running tf2-py==0.25.2 on Ubuntu 20.04. It looks like jump protection was implemented [here](https://github.com/ros2/geometry2/blob/5eee548ac7424f3dd9889ece279db0f3128c58b3/tf2_ros/src/buffer.cpp#LL65C57-L65C69) in the C++ implementation. I'm not really privy to how C++ bindings for python work, but it looks like that little piece of functionality was left out of the python implementation/bindings.

I was able to get this functionality myself with this little snippet:
```
def time_jump_callback(time_jump: TimeJump):
            self.get_logger().warning("Time jump./change detected! Clearing tf buffer.")
            self.tf_buffer.clear()

        threshold = JumpThreshold(min_forward=Duration(seconds=1), min_backward=Duration(seconds=-1), on_clock_change=True)
        self.jump_handle = self.get_clock().create_jump_callback(threshold, post_callback=time_jump_callback)
```
With that alongside my buffer/listener definitions, the node was able to handle time jumps beautifully. Note if you're using this snippet: be careful to hold on to the jump_handle reference, if it goes out of scope and Python deconstructs it you get some very confusing errors.
I'm not sure if it would be better to include this code in the Python implementation, or if it should be properly brought in from the C++ implementation.

erichlf, 2023-06-13:
> I also had this problem running tf2-py==0.25.2 on Ubuntu 20.04. It looks like jump protection was implemented [here](https://github.com/ros2/geometry2/blob/5eee548ac7424f3dd9889ece279db0f3128c58b3/tf2_ros/src/buffer.cpp#LL65C57-L65C69) in the C++ implementation. I'm not privy to how C++ bindings for python work, but it looks like that piece of functionality was left behind.
> 
> I was able to get this functionality myself with this little snippet:
> 
> ```
> def time_jump_callback(time_jump: TimeJump):
>     self.get_logger().warning("Time jump./change detected! Clearing tf buffer.")
>     self.tf_buffer.clear()
> threshold = JumpThreshold(min_forward=Duration(seconds=1), min_backward=Duration(seconds=-1), on_clock_change=True)
> self.jump_handle = self.get_clock().create_jump_callback(threshold, post_callback=time_jump_callback)
> ```
> 
> With that alongside my buffer/listener definitions, tf2 Handled time jumps beautifully. Note if you're using this snippet: be careful to hold on to the jump_handle reference. If it goes out of scope and Python deconstructs it, you get some very confusing errors.
> 
> I'm not sure if it would be better to include this code in the Python implementation, or if it should be properly brought in from the C++ implementation.

This was the same work-around I used.

erichlf, 2023-06-14:
Here is the [PR](https://github.com/ros2/geometry2/pull/608).

CursedRock17, 2024-03-15:
I believe PR #608 fully resolved this, otherwise I can pick this up.

clalancette, 2024-03-15:
Yeah, I'm going to close this out.  Thanks for checking.

closed 2024-03-15 by clalancette