ros2/geometry2 #609: which callback group should be default callback group in transform_listener.py

sorawit112, 2023-06-16:
In branch humble
transform_listener.py have this following lines of code

-- # Default callback group is mutually exclusive, which would prevent waiting for transforms
-- # from another callback in the same group.
-- self.group = ReentrantCallbackGroup()

the comment said default callback group is Mutually Exclusive but the real one is Reentrant so which one is correct?

cmraaron, 2023-07-22:
I believe this comment is saying that the default callback group _would_ be mutually exclusive, if it were not explicitly set.  And therefore explicitly sets it to a `ReentrantCallbackGroup`

clalancette, 2023-07-24:
> I believe this comment is saying that the default callback group _would_ be mutually exclusive, if it were not explicitly set. And therefore explicitly sets it to a `ReentrantCallbackGroup`

Yes, that's correct.

Given that this is a question, I'm going to close this out.  In the future please open questions like this on https://answers.ros.org.