ros2/geometry2 #640: Treating a frame and a transform as the same is unclear without digging into source code

roshambo919, 2023-12-13:
It is my understanding that `tf2`'s use of `PyKDL` to do transformations means that the position and orientation construct a classic transformation matrix:

```
T = [[ M,  p],
     [ 0,  1]]
```

When this type of transformation is applied to a point, the rotation of the point happens first, then the point is translated in the new rotated frame. It took me several hours of digging in the source code to determine this behavior. My personal definition of a reference frame is that it is the opposite order -- translation first then rotation -- as opposed to frame being synonymous with transform.

This is not necessarily a problem with  tf2 BUT caused great headaches trying to understand how to define a frame because the [ros2 tf2 tutorials](https://docs.ros.org/en/foxy/Tutorials/Intermediate/Tf2/Adding-A-Frame-Py.html) only use translations when defining the frame and use the identity quaternion making it ambiguous which order the operations are applied from just the tutorials. 

Moreover, this same behavior is evident when using the function `do_transform_pose`. That function treats the pose as a transform/frame. This means that a transform's position is defined IN THE ROTATED COORDINATE FRAME i.e. AFTER applying the pose's rotation. Personally, I dislike this definition since now the position is coupled with the orientation. Defining pose as first a translation then a rotation would decouple the two. But since that path is unlikely to ever be realized, I would greatly appreciate some enhanced documentation and/or clarification on my misunderstandings.

roshambo919, 2023-12-14:
[This could be a useful discussion point](https://wiki.ros.org/geometry/CoordinateFrameConventions).

```
Transforms are specified in the direction such that they will transform the coordinate frame "frame_id" into "child_frame_id" in tf::StampedTransform and [geometry_msgs/TransformStamped](http://docs.ros.org/en/api/geometry_msgs/html/msg/TransformStamped.html). This is the same transform which will take data from "child_frame_id" into **"frame_id".**
```
I recommend better documentation of `do_transform_point` and `do_transform_pose` to illustrate how to use them properly.

sloretz, 2023-12-26:
@tfoote would you be willing to describe how transformations are done in tf2? 🧇

tfoote, 2023-12-27:
That's not my understanding of the definition of an affine transform. And you're citing the documentation for publishing transforms versus applying transforms. In particular when you change this semantic the value is inverted. And then it will consequently reverse the order of applying the transform.

The API will return you the transform to move a point from the source frame to the target frame. Do transform should apply the transform to the data to move it from the source to target. In the case of a point it is just a left multiply by the transform. This allows you to implement it for more complex data types. 

For more help please ask a question on stack exchange and make sure to provide some concrete examples of what you think should be happening as well as what is not working for you.

closed 2023-12-27 by tfoote