ros2/geometry2 #666: Create (dynamic) Transform Broadcaster

tonynajjar, 2024-04-02:
## Feature request

#### Feature description

Similarly to `static_transform_publisher`, it would be useful to have a `transform_publisher` that publishes to `/tf` on a timer (configurable rate). One use case of this would be to publish map -> odom artificially (e.g. if a global localizer is missing).
I've seen the `static_transform_publisher` being used for this but for me it hasn't worked out for reasons I don't know yet, but besides map -> odom should theoretically be on `/tf` as it's not static

#### Implementation considerations

clalancette, 2024-04-03:
Maybe I'm misunderstanding what you want here, but wouldn't this be as easy as:

```
auto tb = TransformBroadcaster();
auto callback = [&tb]() -> void
{
  geometry_msgs::msg::TransformStamped ts;
  tb->sendTransform(ts);
};
auto timer = node->create_wall_timer(1s, callback);
```

That is, I'm not entirely convinced we need to have core functionality for this; it is easy enough for users to do right now.

tonynajjar, 2024-04-03:
Sorry I wasn't super clear, basically I'm asking for a similar functionality to:

```
    node_static_map_to_odom = Node(
        package="tf2_ros",
        executable="static_transform_publisher",
        arguments=["0", "0", "0", "0", "0", "0", "map", "odom"],
    )
```

Something like:

```
    node_map_to_odom = Node(
        package="tf2_ros",
        executable="transform_publisher",
        parameters=[{"period": 1.0,}], # debatable if parameter or argument
        arguments=["0", "0", "0", "0", "0", "0", "map", "odom"],
    )
```

> but wouldn't this be as easy as

Yes you're right it wouldn't be so hard to create but doesn't mean we can't avoid this work for a lot of ROS users, `static_transform_publisher`, `tf2_echo`, etc... are simple scripts but extremely useful. Besides, it's not **as simple as that**, considering the argument parsing, like is done in `static_transform_broadcaster_program.cpp`

I hope it's clearer what this issue is about?