ros2/geometry2 #669: Wrong Euler convention used for tf2_echo

doisyg, 2024-04-04:
## Bug report

**Required Info:**

- Operating System: Ubuntu 22.04
- Installation type:
  - Latest Iron bin

#### Steps to reproduce issue
Query a 3d transform 
```
ros2 run tf2_ros tf2_echo frame_b frame_a

```

#### Expected behavior

Display RPY Euler rotation 

#### Actual behavior

YPR Euler rotation is displayed

#### Additional information

Due to https://github.com/ros2/geometry2/blob/0ea2ad3ee99d2e3ffcb1f04286d22edfeffb2e1c/tf2_ros/src/tf2_echo.cpp#L159

scott-robotics, 2024-04-14:
Still displayed in RPY convention - just the getter is YPR
https://github.com/ros2/geometry2/blob/0ea2ad3ee99d2e3ffcb1f04286d22edfeffb2e1c/tf2_ros/src/tf2_echo.cpp#L161

doisyg, 2024-04-14:
Yes but then depending on the transformation queried, displaying in the order Roll Pitch Yaw, doesn't make it a RPY convention ;)

doisyg, 2024-04-21:
Closing as I must have been really confused when I opened this issue and I don't see a problem anymore