ros2/geometry2 #703: need libclass_loader.so

XQSwx, 2024-08-16:
Cross compilation
There is a problem

```
Starting >>> tf2_ros
--- stderr: tf2_ros
/buildroot/host/bin/../lib/gcc/aarch64-buildroot-linux-gnu/11.3.0/../../../../aarch64-buildroot-linux-gnu/bin/ld: warning: libclass_loader.so, needed by libstatic_transform_broadcaster_node.so, not found (try using -rpath or -rpath-link)
/buildroot/host/bin/../lib/gcc/aarch64-buildroot-linux-gnu/11.3.0/../../../../aarch64-buildroot-linux-gnu/bin/ld: libstatic_transform_broadcaster_node.so: undefined reference to `class_loader::impl::getCurrentlyLoadingLibraryName[abi:cxx11]()'
/buildroot/host/bin/../lib/gcc/aarch64-buildroot-linux-gnu/11.3.0/../../../../aarch64-buildroot-linux-gnu/bin/ld: libstatic_transform_broadcaster_node.so: undefined reference to `class_loader::impl::getPluginBaseToFactoryMapMapMutex()'
/buildroot/host/bin/../lib/gcc/aarch64-buildroot-linux-gnu/11.3.0/../../../../aarch64-buildroot-linux-gnu/bin/ld: libstatic_transform_broadcaster_node.so: undefined reference to `class_loader::impl::hasANonPurePluginLibraryBeenOpened(bool)'
/buildroot/host/bin/../lib/gcc/aarch64-buildroot-linux-gnu/11.3.0/../../../../aarch64-buildroot-linux-gnu/bin/ld: libstatic_transform_broadcaster_node.so: undefined reference to `class_loader::impl::getCurrentlyActiveClassLoader()'
/buildroot/host/bin/../lib/gcc/aarch64-buildroot-linux-gnu/11.3.0/../../../../aarch64-buildroot-linux-gnu/bin/ld: libstatic_transform_broadcaster_node.so: undefined reference to `class_loader::impl::AbstractMetaObjectBase::setAssociatedLibraryPath(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/buildroot/host/bin/../lib/gcc/aarch64-buildroot-linux-gnu/11.3.0/../../../../aarch64-buildroot-linux-gnu/bin/ld: libstatic_transform_broadcaster_node.so: undefined reference to `class_loader::impl::getFactoryMapForBaseClass(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/buildroot/host/bin/../lib/gcc/aarch64-buildroot-linux-gnu/11.3.0/../../../../aarch64-buildroot-linux-gnu/bin/ld: libstatic_transform_broadcaster_node.so: undefined reference to `class_loader::impl::AbstractMetaObjectBase::AbstractMetaObjectBase(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/buildroot/host/bin/../lib/gcc/aarch64-buildroot-linux-gnu/11.3.0/../../../../aarch64-buildroot-linux-gnu/bin/ld: libstatic_transform_broadcaster_node.so: undefined reference to `class_loader::impl::AbstractMetaObjectBase::addOwningClassLoader(class_loader::ClassLoader*)'
collect2: error: ld returned 1 exit status
gmake[2]: *** [CMakeFiles/static_transform_publisher.dir/build.make:197: static_transform_publisher] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:226: CMakeFiles/static_transform_publisher.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< tf2_ros [1.42s, exited with code 2]
```

By adding the following files, you can proceed
```
diff --git a/tf2_ros/CMakeLists.txt b/tf2_ros/CMakeLists.txt
index 7f15ed83..b5cc656c 100644
--- a/tf2_ros/CMakeLists.txt
+++ b/tf2_ros/CMakeLists.txt
@@ -19,7 +19,7 @@ find_package(rclcpp_action REQUIRED)
 find_package(rclcpp_components REQUIRED)
 find_package(tf2 REQUIRED)
 find_package(tf2_msgs REQUIRED)
-
+find_package(class_loader REQUIRED)
 # tf2_ros library
 add_library(${PROJECT_NAME} SHARED
   src/buffer.cpp
@@ -38,6 +38,7 @@ target_link_libraries(${PROJECT_NAME} PUBLIC
   ${geometry_msgs_TARGETS}
   message_filters::message_filters
   rclcpp::rclcpp
+  class_loader::class_loader
   rclcpp_action::rclcpp_action
   tf2::tf2
   ${tf2_msgs_TARGETS})
```

clalancette, 2024-08-19:
Adding in a dependency on `class_loader` shouldn't be required here, as `tf2_ros` does not directly depend on it.  So I don't think that is the correct fix.

That said, it may be the case that something that `tf2_ros` deos depend on (for example, `rclcpp_components`) isn't properly exporting a dependency that it should be.

What version of ROS 2 are you trying to cross-compile?

clalancette, 2024-12-04:
No response in a long time, so closing this out.  Feel free to reopen and provide the necessary information.

closed 2024-12-04 by clalancette